cartographer-project/cartographer_ros #1486: VLP16 and IMU tuning issue

ikhann, 2020-06-25:
Hello, 
I am tuning 3D Сartographer for my data, but there is a big drift during trajectory build, especially in the vertical channel, I can't solve it, can you help me please? My sensors: Lidar VLP-16, IMU.

**RVIZ**

[image]

**VALIDATE**

[image]

**The configuration files:**

1. [launch file](https://gist.github.com/ikhann/321598b3e3d21787ff65a3da3e46f935)

2. [lua file](https://gist.github.com/ikhann/a25bc47e2e8a7b91f2cf190a7d1dfc2c)

2. [urdf file](https://gist.github.com/ikhann/569a644756b9447ba268c7d053a0e6b3)

3. [rviz file](https://gist.github.com/ikhann/4509654771d9c6430290045352f68c2e)

and the bag file is [HERE](https://drive.google.com/file/d/1WZVEAo9GjAqeXOl04e4wdIFlfs-JO-zN/view?usp=sharing).

**PROJECTION XRAY XZ**

[image]

**PROJECTION XRAY XY**

[image]

Thanks for your help and advises!

ikhann, 2020-06-25:
> 
> Thanks for your help and advises!

Especially from @SirVer :pray:

georgflick, 2020-08-17:
For me it looks like the transformation between imu and the VLP16 is off. Have you checked and calibrated the extrinsics between them?

Another idea is to use the imu extrincis estimator:
`POSE_GRAPH.optimization_problem.use_online_imu_extrinsics_in_3d = true / false` to enable/disable the online imu estimation. Maybe worth a try.

If you don't have any changes in z direction you could also fix the z values with:
`POSE_GRAPH.optimization_problem.fix_z_in_3d = true`

ikhann, 2020-08-17:
> For me it looks like the transformation between imu and the VLP16 is off. Have you checked and calibrated the extrinsics between them?
>
> Another idea is to use the imu extrincis estimator: [...]

To be honest, since I posted my issue I reached sufficient results. 
Specifically, I got huge improvements in building map when tuned a parameter `TRAJECTORY_BUILDER_3D.num_accumulated_range_data` from 1 to 3.

Some screenshots:
[image]
[image]

My next problem:

On the picture below you can see a road surface on TopDown view:
[image]

I downsampled point cloud using 15 cm Voxel Filter.
Ideally, the road should be a set of points in a single layer, but here on the view from the top are 2-3 layers of the dots and it means that I have 30-45 cm layer of points in high (i got 2-3 dots when downsampled cloud with Voxel filter 15 cm).

So could you help how to build road surface in one layer?