cartographer-project/cartographer_ros #1487: Error when building cartographer:ros_log_sink.cc

FighterEVO, 2020-06-30:
In function `cartographer_ros::ScopedRosLogSink::send(int, char const*, char const*, int, tm const*, char const*, unsigned long)':
ros_log_sink.cc:(.text+0x144): undefined reference to `google::LogSink::ToString[abi:cxx11](int, char const*, int, tm const*, char const*, unsigned long, int)'

I have never come across this problem in previous version. Any help would be greatly appreciated!

FighterEVO, 2020-07-01:
I solved this problem by upgradding the version of protobuf, thanks!

closed 2020-07-01 by FighterEVO

bill4u, 2020-11-30:
> I solved this problem by upgradding the version of protobuf, thanks!

would you please share your version of protobuf? I met the same problem.THX!