cartographer-project/cartographer_ros #1492: Cartographer_ros compiling error

nyangshawbin, 2020-07-11:
Hi!

I followed the installation guide on [google cartographer](https://google-cartographer-ros.readthedocs.io/en/latest/compilation.html) successfully, but realized the [pbstream_trajectories_to_rosbag](https://github.com/cartographer-project/cartographer_ros/blob/master/cartographer_ros/cartographer_ros/dev/pbstream_trajectories_to_rosbag_main.cc) node is missing. 

As such, I replaced the cartographer_ros package from the installation with this repo. However, there seems to be a error compiling with regarding "absl". Following is a snippet of the log:
```
CMake Error at CMakeLists.txt:57 (find_package):
  By not providing "Findabsl.cmake" in CMAKE_MODULE_PATH this project has
  asked CMake to find a package configuration file provided by "absl", but
  CMake did not find one.

  Could not find a package configuration file provided by "absl" with any of
  the following names:

    abslConfig.cmake
    absl-config.cmake

  Add the installation prefix of "absl" to CMAKE_PREFIX_PATH or set
  "absl_DIR" to a directory containing one of the above files.  If "absl"
  provides a separate development package or SDK, be sure it has been
  installed.

-- Configuring incomplete, errors occurred!
See also "/home/shaw/cartographer_ws/build_isolated/cartographer_ros/CMakeFiles/CMakeOutput.log".
See also "/home/shaw/cartographer_ws/build_isolated/cartographer_ros/CMakeFiles/CMakeError.log".
FAILED: build.ninja 
/usr/bin/cmake -H/home/shaw/cartographer_ws/src/cartographer_ros/cartographer_ros -B/home/shaw/cartographer_ws/build_isolated/cartographer_ros
ninja: error: rebuilding 'build.ninja': subcommand failed
<== Failed to process package 'cartographer_ros': 
  Command '['/home/shaw/cartographer_ws/install_isolated/env.sh', 'ninja', 'build.ninja']' returned non-zero exit status 1
```

Or is there a simpler way of me adding the missing node instead of replacing the entire package. Thank you in advance!

ayushgargdroid, 2020-07-15:
Hi @nyangshawbin 

I got the same error. I solved it by running `install_abseil.sh` file in the original cartographer project. It is in the `scripts` directory.

nyangshawbin, 2020-07-15:
Hi @ayushgargdroid , yes! That solved it. Thanks so much for your help, really appreciate it!!

closed 2020-07-15 by nyangshawbin