cartographer-project/cartographer_ros #1503: WARN Could not compute submap fading: "base_link" passed to lookupTransform argument source_frame does not exist.

darthShana, 2020-08-11:
Hi im new to ROS and cartographer..
Im running a RPLIDAR with https://github.com/robopeak/rplidar_ros

and then using cartographer_ros.. 
Things look like they are working correctly. I can see on rviz my path through the map.. And i have verified my laser frame translation with respect to map looks correct.. 

However i get this warning repeatedly about the missing base_link frame

`[ WARN] [1597135213.244415725]: Could not compute submap fading: "base_link" passed to lookupTransform argument source_frame does not exist.`

Is that supposed to be created by the cartographer?
or should submap fading be using a different frame?

my frames look like 
map -> odom -> laser

my config file is

```
include "map_builder.lua"
include "trajectory_builder.lua"

options = {
 map_builder = MAP_BUILDER,
 trajectory_builder = TRAJECTORY_BUILDER,
 map_frame = "map",
 tracking_frame = "laser",
 published_frame = "laser",
 odom_frame = "odom",
 provide_odom_frame = true,
 publish_frame_projected_to_2d = false,
 use_odometry = false,
 use_nav_sat = false,
 use_landmarks = false,
 num_laser_scans = 1,
 num_multi_echo_laser_scans = 0,
 num_subdivisions_per_laser_scan = 1,
 num_point_clouds = 0,
 lookup_transform_timeout_sec = 0.2,
 submap_publish_period_sec = 0.3,
 pose_publish_period_sec = 5e-3,
 trajectory_publish_period_sec = 30e-3,
 rangefinder_sampling_ratio = 1.,
 odometry_sampling_ratio = 1.,
 fixed_frame_pose_sampling_ratio = 1.,
 imu_sampling_ratio = 1.,
 landmarks_sampling_ratio = 1.,
}

```
I have not changed any other files
my startup looks like
```
oslaunch cartographer_ros myrobot.launch 
... logging to /home/khadas/.ros/log/28bd56f2-dbae-11ea-8931-c8631470c8c2/roslaunch-Khadas-8454.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://Khadas:37655/

SUMMARY
========

PARAMETERS
 * /rosdistro: melodic
 * /rosversion: 1.14.6

NODES
  /
    cartographer_node (cartographer_ros/cartographer_node)
    cartographer_occupancy_grid_node (cartographer_ros/cartographer_occupancy_grid_node)
    rviz (rviz/rviz)

ROS_MASTER_URI=http://localhost:11311

process[cartographer_node-1]: started with pid [8469]
process[cartographer_occupancy_grid_node-2]: started with pid [8470]
[ INFO] [1597135197.473732176]: I0811 08:39:57.000000  8469 configuration_file_resolver.cc:41] Found '/home/khadas/catkin_ws/install_isolated/share/cartographer_ros/configuration_files/my_robot.lua' for 'my_robot.lua'.
[ INFO] [1597135197.476959421]: I0811 08:39:57.000000  8469 configuration_file_resolver.cc:41] Found '/home/khadas/catkin_ws/install_isolated/share/cartographer/configuration_files/map_builder.lua' for 'map_builder.lua'.
[ INFO] [1597135197.477104504]: I0811 08:39:57.000000  8469 configuration_file_resolver.cc:41] Found '/home/khadas/catkin_ws/install_isolated/share/cartographer/configuration_files/map_builder.lua' for 'map_builder.lua'.
[ INFO] [1597135197.477294879]: I0811 08:39:57.000000  8469 configuration_file_resolver.cc:41] Found '/home/khadas/catkin_ws/install_isolated/share/cartographer/configuration_files/pose_graph.lua' for 'pose_graph.lua'.
[ INFO] [1597135197.477380462]: I0811 08:39:57.000000  8469 configuration_file_resolver.cc:41] Found '/home/khadas/catkin_ws/install_isolated/share/cartographer/configuration_files/pose_graph.lua' for 'pose_graph.lua'.
[ INFO] [1597135197.477708920]: I0811 08:39:57.000000  8469 configuration_file_resolver.cc:41] Found '/home/khadas/catkin_ws/install_isolated/share/cartographer/configuration_files/trajectory_builder.lua' for 'trajectory_builder.lua'.
[ INFO] [1597135197.477807920]: I0811 08:39:57.000000  8469 configuration_file_resolver.cc:41] Found '/home/khadas/catkin_ws/install_isolated/share/cartographer/configuration_files/trajectory_builder.lua' for 'trajectory_builder.lua'.
[ INFO] [1597135197.477952878]: I0811 08:39:57.000000  8469 configuration_file_resolver.cc:41] Found '/home/khadas/catkin_ws/install_isolated/share/cartographer/configuration_files/trajectory_builder_2d.lua' for 'trajectory_builder_2d.lua'.
[ INFO] [1597135197.478033419]: I0811 08:39:57.000000  8469 configuration_file_resolver.cc:41] Found '/home/khadas/catkin_ws/install_isolated/share/cartographer/configuration_files/trajectory_builder_2d.lua' for 'trajectory_builder_2d.lua'.
[ INFO] [1597135197.478299044]: I0811 08:39:57.000000  8469 configuration_file_resolver.cc:41] Found '/home/khadas/catkin_ws/install_isolated/share/cartographer/configuration_files/trajectory_builder_3d.lua' for 'trajectory_builder_3d.lua'.
[ INFO] [1597135197.478380252]: I0811 08:39:57.000000  8469 configuration_file_resolver.cc:41] Found '/home/khadas/catkin_ws/install_isolated/share/cartographer/configuration_files/trajectory_builder_3d.lua' for 'trajectory_builder_3d.lua'.
process[rviz-3]: started with pid [8487]
[ INFO] [1597135197.493227979]: I0811 08:39:57.000000  8469 submap_2d.cc:187] Added submap 1
[ INFO] [1597135197.493461895]: I0811 08:39:57.000000  8469 map_builder_bridge.cc:130] Added trajectory with ID '0'.
libEGL warning: DRI2: failed to authenticate
[ INFO] [1597135197.787461563]: I0811 08:39:57.000000  8469 ordered_multi_queue.cc:172] All sensor data for trajectory 0 is available starting at '637327319977858154'.
[ INFO] [1597135197.787623521]: I0811 08:39:57.000000  8469 local_trajectory_builder_2d.cc:138] Extrapolator is still initializing.
[ WARN] [1597135199.000871508]: Could not compute submap fading: "base_link" passed to lookupTransform argument source_frame does not exist. 
```

any help would be appreciated

darthShana, 2020-08-11:
ok had a little dig through the code and figured it out..
demo_2d.rviz
has a property Tracking Frame that need to be changed..

closed 2020-08-11 by darthShana

csufangyu, 2020-12-09:
hi I get the same error,could you tell me how change  Tracking Frame?

johuber, 2020-12-09:
> hi I get the same error,could you tell me how change Tracking Frame?

change the parameter `tracking_frame` in your parameter file

csufangyu, 2020-12-09:
> > hi I get the same error,could you tell me how change Tracking Frame?
> 
> change the parameter `tracking_frame` in your parameter file

What is this parameter set to？

johuber, 2020-12-09:
> > > hi I get the same error,could you tell me how change Tracking Frame?
> > 
> > 
> > change the parameter `tracking_frame` in your parameter file
> 
> What is this parameter set to？

That depends on your own setup. Please familiarize yourself with transform in ROS and it should become clear after that. E.g. here there are tons of tutorials to find:
http://wiki.ros.org/navigation/Tutorials/RobotSetup/TF

csufangyu, 2020-12-09:
thanks very much！方宇-------- 原始邮件 --------发件人： Jonathan Huber <notifications@github.com>日期： 2020年12月9日周三 傍晚5:06收件人： cartographer-project/cartographer_ros <cartographer_ros@noreply.github.com>抄送： universe <csufangyu@163.com>, Comment <comment@noreply.github.com>主    题： Re: [cartographer-project/cartographer_ros] WARN Could not compute submap fading: "base_link" passed to lookupTransform argument source_frame does not exist. (#1503)



hi I get the same error,could you tell me how change Tracking Frame?

change the parameter tracking_frame in your parameter file

What is this parameter set to？

That depends on your own setup. Please familiarize yourself with transform in ROS and it should become clear after that. E.g. here there are tons of tutorials to find:
http://wiki.ros.org/navigation/Tutorials/RobotSetup/TF

weihaoysgs, 2022-05-09:
thanks your answer, i have the same problem and solve it.

Hanoch666, 2022-06-10:
use_odometry = false

annanyatyagi12, 2023-04-28:
has anyone figured out how to solve this?