cartographer-project/cartographer_ros #1505: pure localization usage

darthShana, 2020-08-20:
HI 
Im trying to figure how to get localization working, im following the documentation on
https://google-cartographer-ros.readthedocs.io/en/latest/going_further.html

by adding 
`TRAJECTORY_BUILDER.pure_localization_trimmer = {
    max_submaps_to_keep = 3,
}`

however when i run 
`roslaunch cartographer_ros room_2d_localization.launch load_state_filename:=/home/darthshana/bagfiles/2020-08-20-08-40-58.bag.pbstream bag_filename:=/home/darthshana/bagfiles/2020-08-20-08-42-40.bag`

i get the error
`[FATAL] [1597909932.178603154]: F0820 19:52:12.000000 10348 lua_parameter_dictionary.cc:410] Check failed: 1 == reference_counts_.count(key) (1 vs. 0) Key 'pure_localization_trimmer' was used the wrong number of times.
`
I also tried without the config line, then i get lots of
`E0820 19:28:58.919713  6169 map_builder_bridge.cc:175] Requested submap 6 from trajectory 0 but it does not exist: maybe it has been trimmed.
[ERROR] [1597908538.919817917, 1597869816.582156400]: E0820 19:28:58.000000  6169 map_builder_bridge.cc:175] Requested submap 6 from trajectory 0 but it does not exist: maybe it has been trimmed.`

And im also not clear where i can i find the localised pose of the robot? perhaps its not being published due to the error

MichaelGrupp, 2020-09-01:
This can happen if you use RViz and it tries to visualize a submap that doesn't exist anymore.

The pose of the robot is published on the `/tf` topic depending on your frame configuration.

closed 2020-10-14 by MichaelGrupp

IqbalBintangsyah, 2024-05-17:
Hello, I also have this same problem with ROS2-Humble

What does it mean by a submap doesn't exist anymore? How do I stop it from happening?

Thank you