cartographer-project/cartographer_ros #1507: Cartographer multi-session mapping

sgr-gpt, 2020-08-21:
Hello everyone,

I was exploring the cartographer-package and wanted to know if it has support for multi-session mapping. I would really appreciate some help here.

Thanks

MichaelGrupp, 2020-09-01:
Can you elaborate a bit more what you mean with multi-session mapping?

sgr-gpt, 2020-09-01:
The task I wish to address is suppose I map a part of a room today using cartographer and store it and kill all the running processes then would I be able to continue the mapping of the remaining room the next day and combine the two maps (online/offline both methods are considerable)? Would this be possible through the ros package provided by cartographer?

MichaelGrupp, 2020-09-01:
You can serialize the state to a .pbstream file (`/write_state` service) and the next day load this file again via `--load_state_filename` argument of the node and continue mapping, potentially using the `/start_trajectory` service to set a reasonable start pose. Alternatively, the offline node also supports processing multiple bagfiles, but this is probably not suitable if there is a large gap between mappings.

sgr-gpt, 2020-09-07:
Thank you @MichaelGrupp this approach worked as desired.

closed 2020-09-07 by sgr-gpt