cartographer-project/cartographer_ros #1513: about  Occupancy grid maps color (black/white/gray) at rviz. (3D slam with 2 Lidar and IMU)

sizsJEon, 2020-09-09:
I'm currently test cartographer_ros in mobile platform with 2 Lidars and 1 6DOF IMU.
(Lidar : Horizontal and Vertical to Ground)

Based on Deutch museum example's, I made my own urdf and configuration files.
And could run cartographer.

but After some test, I found that, 
Unlike 2d cartographer, 3d cartographer's white area's are always found in above plane(ceiling, high wall, trees)
[image]

My question is, 
1. in 3D cartograper, what is the meaning of white and gray zone?

2. If the white area coming out of the high place is due to the wrong setting, how could I fix it? (I want ground to be white area)
  I've already tried inverting IMU, vertical Lidar (roll +-180). but result was same.

MichaelGrupp, 2020-10-14:
The 3D grid is converted into an occupancy grid image for visualization. The actual conversion takes place here: https://github.com/cartographer-project/cartographer/blob/3b511aa1bafaf3cd80890927bf7122a33e5f13a3/cartographer/mapping/3d/submap_3d.cc#L148
Especially `ComputePixelValues` is relevant here.

MichaelGrupp, 2020-10-14:
More information also here: https://github.com/cartographer-project/cartographer_ros/issues/330#issuecomment-300385597