cartographer-project/cartographer_ros #1515: cartographer 3d mapping in ROS2

rjshim, 2020-09-14:
Hi,

I wonder if this repository supports 3d mapping in ROS2. Since Dashing does not support XML format launch files, I think [this](https://github.com/cartographer-project/cartographer_ros/blob/dashing/cartographer_ros/launch/backpack_3d.launch) has not been tested but just uploaded without testing. and It even looks just a copy-paste of the master branch.

Ryan

MichaelGrupp, 2020-10-14:
ROS2 is not officially supported by cartographer_ros yet, but there are plans to do so. We are unfortunately a bit slow with this at the moment as we are currently reviving this repository again as a community-based project. There is a pull request, but it is currently just a snapshot of a very outdated fork: #1488. The work that needs to be done is to apply similar changes to the current ROS1 master, or to integrate the changes that were made in ROS1 to the old ROS2 branch.

But if you want to accelerate this development of porting to ROS2, you're more than welcome to participate. Let me know if you have questions regarding that.

Serafadam, 2020-10-15:
I have been using [ros2 cartographer repository](https://github.com/ros2/cartographer_ros) with some success

JaehyunShim, 2020-10-16:
@Serafadam @MichaelGrupp 

Thank you. maybe I use the repository from the ros2 community. It looks well maintained and updated. I will close this issue.

closed 2020-10-16 by JaehyunShim

MichaelGrupp, 2020-10-16:
Be aware that the ros2 repository is now 2.5 years behind upstream. But it should compile. As I said, this is the reason we and the OSRF want to make a new version here.

JaehyunShim, 2020-10-16:
@MichaelGrupp 
Thank you for your comment. I will take a look at their repository and see what I can contribute here.

Jaehyun

MichaelGrupp, 2020-10-16:
Awesome, it would be great to get help for ROS2. If you have any questions let me know.

Serafadam, 2020-10-26:
I think I'd also like to contribute to a newer port, but I have some questions regarding this first, should I create a new issue for that? @MichaelGrupp

MichaelGrupp, 2020-10-26:
@Serafadam Cool! You can comment here: #1536 I created a pinned issue so this doesn't get lost. We as maintainers will discuss some things this week and the whole thing is in a very early stage, but as I said I would be more than happy to involve people on the way.