cartographer-project/cartographer_ros #1526: Mapping Issue

LuisIGV2020, 2020-10-14:
Hi,

I am trying to run cartographer_ros slam on ubuntu 16.04 and using ros noetic. I have built and installed the cartographer as well as cartographer_ros from this github and am running into some errors. To simulate the program I am using Turtlebot3 waffle pi and for the mapping I am using gazebo and rviz. Below are some images I am having as well as the rqt_tf_tree. Any and all help will be appreciated since it is for a university senior design!

[image]

[image]

[image]

3473f, 2020-10-14:
Can you share your .lua and launch files?

LuisIGV2020, 2020-10-15:
[turtlebot3_cartographer.pdf](https://github.com/cartographer-project/cartographer_ros/files/5388178/turtlebot3_cartographer.pdf)

[turtlebot3_lds_2d.pdf](https://github.com/cartographer-project/cartographer_ros/files/5388179/turtlebot3_lds_2d.pdf)

The first image is the launch files and the second is the .lua files. Thank you for your help

LuisIGV2020, 2020-10-15:
(This is the launch file)

<launch>
  
  <arg name="model" default="$(env TURTLEBOT3_MODEL)" doc="model type [burger, waffle, waffle_pi]"/>
  <arg name="configuration_basename" default="turtlebot3_lds_2d.lua"/>

  
  <include file="$(find turtlebot3_navigation)/launch/move_base.launch">
    <arg name="model" value="$(arg model)" />
  </include>

  
  <node pkg="cartographer_ros" type="cartographer_node" name="cartographer_node" 
        args="-configuration_directory $(find turtlebot3_slam)/config
              -configuration_basename $(arg configuration_basename)"
        output="screen">
    <remap from="/base_footprint" to="/flat_imu"/>
    
  </node>

  
  <node pkg="cartographer_ros" type="cartographer_occupancy_grid_node"
        name="cartographer_occupancy_grid_node" 
        args="-resolution 0.05" />

  
  <node pkg="turtlebot3_slam" type="flat_world_imu_node" name="flat_world_imu_node" output="screen">
    <remap from="imu_in" to="/imu" />
    <remap from="imu_out" to="/flat_imu" />
  </node>
</launch>

-------------------------------------------------------------------------------------------------------------------------------------
(This is the .lua file)
-- Copyright 2016 The Cartographer Authors
--
-- Licensed under the Apache License, Version 2.0 (the "License");
-- you may not use this file except in compliance with the License.
-- You may obtain a copy of the License at
--
--      http://www.apache.org/licenses/LICENSE-2.0
--
-- Unless required by applicable law or agreed to in writing, software
-- distributed under the License is distributed on an "AS IS" BASIS,
-- WITHOUT WARRANTIES OR CONDITIONS OF ANY KIND, either express or implied.
-- See the License for the specific language governing permissions and
-- limitations under the License.

include "map_builder.lua"
include "trajectory_builder.lua"

options = {
  map_builder = MAP_BUILDER,
  trajectory_builder = TRAJECTORY_BUILDER,
  map_frame = "map",
  tracking_frame = "base_footprint", -- imu_link, If you are using gazebo, use 'base_footprint' (libgazebo_ros_imu's bug)
  published_frame = "odom",
  odom_frame = "odom",
  provide_odom_frame = false,
  publish_frame_projected_to_2d = false,
  use_odometry = true,
  use_nav_sat = false,
  use_landmarks = false,
  num_laser_scans = 1,
  num_multi_echo_laser_scans = 0,
  num_subdivisions_per_laser_scan = 1,
  num_point_clouds = 0,
  lookup_transform_timeout_sec = 0.2,
  submap_publish_period_sec = 0.3,
  pose_publish_period_sec = 5e-3,
  trajectory_publish_period_sec = 30e-3,
  rangefinder_sampling_ratio = 1.,
  odometry_sampling_ratio = 0.1,
  fixed_frame_pose_sampling_ratio = 1.,
  imu_sampling_ratio = 0.1,
  landmarks_sampling_ratio = 1.,
}

MAP_BUILDER.use_trajectory_builder_2d = true

TRAJECTORY_BUILDER_2D.min_range = 0.1
TRAJECTORY_BUILDER_2D.max_range = 3.5
TRAJECTORY_BUILDER_2D.missing_data_ray_length = 3.
TRAJECTORY_BUILDER_2D.use_imu_data = true
TRAJECTORY_BUILDER_2D.use_online_correlative_scan_matching = true
TRAJECTORY_BUILDER_2D.motion_filter.max_angle_radians = math.rad(0.1)

POSE_GRAPH.constraint_builder.min_score = 0.65
POSE_GRAPH.constraint_builder.global_localization_min_score = 0.7

return options

LuisIGV2020, 2020-10-15:
<launch>
  
  <arg name="model" default="$(env TURTLEBOT3_MODEL)" doc="model type [burger, waffle, waffle_pi]"/>
  <arg name="configuration_basename" default="turtlebot3_lds_2d.lua"/>

  
  <include file="$(find turtlebot3_navigation)/launch/move_base.launch">
    <arg name="model" value="$(arg model)" />
  </include>

  
  <node pkg="cartographer_ros" type="cartographer_node" name="cartographer_node" 
        args="-configuration_directory $(find turtlebot3_slam)/config
              -configuration_basename $(arg configuration_basename)"
        output="screen">
    <remap from="/base_footprint" to="/flat_imu"/>
    
  </node>

  
  <node pkg="cartographer_ros" type="cartographer_occupancy_grid_node"
        name="cartographer_occupancy_grid_node" 
        args="-resolution 0.05" />

  
  <node pkg="turtlebot3_slam" type="flat_world_imu_node" name="flat_world_imu_node" output="screen">
    <remap from="imu_in" to="/imu" />
    <remap from="imu_out" to="/flat_imu" />
  </node>
</launch>

LuisIGV2020, 2020-10-15:
`

<launch>
  
  <arg name="model" default="$(env TURTLEBOT3_MODEL)" doc="model type [burger, waffle, waffle_pi]"/>
  <arg name="configuration_basename" default="turtlebot3_lds_2d.lua"/>

  
  <include file="$(find turtlebot3_navigation)/launch/move_base.launch">
    <arg name="model" value="$(arg model)" />
  </include>

  
  <node pkg="cartographer_ros" type="cartographer_node" name="cartographer_node" 
        args="-configuration_directory $(find turtlebot3_slam)/config
              -configuration_basename $(arg configuration_basename)"
        output="screen">
    <remap from="/base_footprint" to="/flat_imu"/>
    
  </node>

  
  <node pkg="cartographer_ros" type="cartographer_occupancy_grid_node"
        name="cartographer_occupancy_grid_node" 
        args="-resolution 0.05" />

  
  <node pkg="turtlebot3_slam" type="flat_world_imu_node" name="flat_world_imu_node" output="screen">
    <remap from="imu_in" to="/imu" />
    <remap from="imu_out" to="/flat_imu" />
  </node>
</launch>`

LuisIGV2020, 2020-10-15:
I was trying to upload the text but it kept coming out invisible

LuisIGV2020, 2020-10-15:
-- Copyright 2016 The Cartographer Authors
--
-- Licensed under the Apache License, Version 2.0 (the "License");
-- you may not use this file except in compliance with the License.
-- You may obtain a copy of the License at
--
--      http://www.apache.org/licenses/LICENSE-2.0
--
-- Unless required by applicable law or agreed to in writing, software
-- distributed under the License is distributed on an "AS IS" BASIS,
-- WITHOUT WARRANTIES OR CONDITIONS OF ANY KIND, either express or implied.
-- See the License for the specific language governing permissions and
-- limitations under the License.

include "map_builder.lua"
include "trajectory_builder.lua"

options = {
  map_builder = MAP_BUILDER,
  trajectory_builder = TRAJECTORY_BUILDER,
  map_frame = "map",
  tracking_frame = "base_footprint", -- imu_link, If you are using gazebo, use 'base_footprint' (libgazebo_ros_imu's bug)
  published_frame = "odom",
  odom_frame = "odom",
  provide_odom_frame = false,
  publish_frame_projected_to_2d = false,
  use_odometry = true,
  use_nav_sat = false,
  use_landmarks = false,
  num_laser_scans = 1,
  num_multi_echo_laser_scans = 0,
  num_subdivisions_per_laser_scan = 1,
  num_point_clouds = 0,
  lookup_transform_timeout_sec = 0.2,
  submap_publish_period_sec = 0.3,
  pose_publish_period_sec = 5e-3,
  trajectory_publish_period_sec = 30e-3,
  rangefinder_sampling_ratio = 1.,
  odometry_sampling_ratio = 0.1,
  fixed_frame_pose_sampling_ratio = 1.,
  imu_sampling_ratio = 0.1,
  landmarks_sampling_ratio = 1.,
}

MAP_BUILDER.use_trajectory_builder_2d = true

TRAJECTORY_BUILDER_2D.min_range = 0.1
TRAJECTORY_BUILDER_2D.max_range = 3.5
TRAJECTORY_BUILDER_2D.missing_data_ray_length = 3.
TRAJECTORY_BUILDER_2D.use_imu_data = true
TRAJECTORY_BUILDER_2D.use_online_correlative_scan_matching = true
TRAJECTORY_BUILDER_2D.motion_filter.max_angle_radians = math.rad(0.1)

POSE_GRAPH.constraint_builder.min_score = 0.65
POSE_GRAPH.constraint_builder.global_localization_min_score = 0.7

return options

3473f, 2020-10-16:
They look fine yo me. Is your laser scanner seeing anything and not just returning invalid measurements (less than range_min or more than range_max in the LaserScan message? Try to visualize it in rviz or better run `rostopic echo scan` while your simulation is running and see which values are being sent.

LuisIGV2020, 2020-10-23:
When I run the command this pops up 
[image]

Another thing that happens when I run the command roslaunch turtlebot3_slam turtlebot3_slam.launch slam_methods:=cartographer

[image]

LuisIGV2020, 2020-10-23:
Sorry for responding back so late

3473f, 2020-10-25:
The laser scan message looks fine. This seems to be similar to #278. I am not using a turtlebot, so I cannot reproduce the error myself. 

I have just noticed that in the file you attached, the tracking frame is set to `imu_link`, while in your comment it is set to `base_footprint` which is what it should be when using Gazebo. Make sure you are loading the correct config file.

LuisIGV2020, 2020-10-29:
Ok, thank you! I did not see that

LuisIGV2020, 2020-11-19:
Got everything working, Thank you for the help!

closed 2020-11-19 by LuisIGV2020