cartographer-project/cartographer_ros #1527: Help with tuning cartographer for UAV mapping

fennerh, 2020-10-16:
Hi,

We have an Ouster OS-1 128bit Lidar mounted to a quadcopter, with a nadir field of view (i.e. the lidar is looking directly down and set to only collect 180 degree point clouds of the ground). Currently struggling with cartographer tuning in order to output complete point clouds.

Example flight plan is a simple out and back rectangle along a line of trees, flying at 30m altitude and 3m/s flight speed. Initial attempts at SLAM processing produce curved flight paths, particualrly on the return track, as well as poor final point clouds as shown by the images below.

X and Y viewpoint:
[image]

X and Z viewpoint:
[image]

Y and Z viewpoint:
[image]

I have had a go at tuning myself, follwoing the two cartogrpaher_ros tuning articles (here: https://google-cartographer-ros.readthedocs.io/en/latest/tuning.html and here: https://google-cartographer-ros.readthedocs.io/en/latest/algo_walkthrough.html) with limited success, mainly due to the impact of the different settings on the SLAM results being unclear to me.

I have attached copies of the rosbag_validate output, .launch file, .lua file and .urdf. 
[rosbag_vlaidate.txt](https://github.com/cartographer-project/cartographer_ros/files/5390908/rosbag_vlaidate.txt)
[offline_ouster_3d.txt](https://github.com/cartographer-project/cartographer_ros/files/5390912/offline_ouster_3d.txt)
[os.txt](https://github.com/cartographer-project/cartographer_ros/files/5390915/os.txt)
[ouster_3d..txt](https://github.com/cartographer-project/cartographer_ros/files/5390916/ouster_3d.txt)

The .bag file can downloaded from here: https://drive.google.com/file/d/1AL35WjR0XZNNgjp2L6SWQvyohJS3g-LD/view?usp=sharing

I would really appreciate any guidance, suggestions or assitance anyone can offer to improve the SLAM processing results from cartographer_ros.

Many thanks

fawadahm, 2020-11-12:
Did you have any luck doing this?

fennerh, 2020-11-13:
Hi,

I have mananged to have some success recently improving cartographer point cloud outputs.

I found the below settings made a significant impact on results:

TRAJECTORY_BUILDER_3D.cerese_scan_matcher.translation_weight = 1
TRAJECTORY_BUILDER_3D.cerese_scan_matcher.rotation_weight = 1

TRAJECTORY_BUILDER_3D.submaps.num_range_data = 1e3

POSE_GRAPH.optimize_every_n_nodes = 1e3

These adjustments fixed the skewed trajectories both in the XY and Z axis; and have found these settings to produce consistent results with larger UAV surverys we have since conducted.

Not 100% confident the tuning I have done is appropirate etc. but seems to be working so far. I welcome any suggestions, advice you or anyone lese might have.