cartographer-project/cartographer_ros #1528: Build error

ArtlyStyles, 2020-10-20:
When I build the ROS workspace, I got the following error:

=> ninja -j16 -l16 in '/home/gcong/catkin_ws/build_isolated/cartographer/install'
[18/278] Building CXX object CMakeFiles/cartographer.mapping.3d.hybrid_grid_test.dir/cartographer/mapping/3d/hybrid_grid_test.cc.o
FAILED: CMakeFiles/cartographer.mapping.3d.hybrid_grid_test.dir/cartographer/mapping/3d/hybrid_grid_test.cc.o 
/usr/bin/c++  -DBOOST_ALL_NO_LIB -DBOOST_IOSTREAMS_DYN_LINK -DGFLAGS_IS_A_DLL=0 -I/home/gcong/catkin_ws/src/cartographer/cartographer -I. -I/home/gcong/catkin_ws/src/cartographer -isystem /usr/src/googletest/googlemock/gtest/include -isystem /usr/include/eigen3 -isystem /home/gcong/catkin_ws/install_isolated/include -isystem /usr/include/lua5.2 -isystem /usr/include/cairo -isystem /usr/include/glib-2.0 -isystem /usr/lib/x86_64-linux-gnu/glib-2.0/include -isystem /usr/include/pixman-1 -isystem /usr/include/uuid -isystem /usr/include/freetype2 -isystem /usr/include/libpng16 -isystem /usr/src/googletest/googlemock/include -isystem /usr/src/googletest/googlemock -isystem /usr/src/googletest/googletest/include -isystem /usr/src/googletest/googletest -O3 -DNDEBUG    -pthread -std=c++11 -fPIC  -Wall -Wpedantic -Werror=format-security -Werror=missing-braces -Werror=reorder -Werror=return-type -Werror=switch -Werror=uninitialized -O3 -DNDEBUG -pthread -std=c++11 -fPIC  -Wall -Wpedantic -Werror=format-security -Werror=missing-braces -Werror=reorder -Werror=return-type -Werror=switch -Werror=uninitialized -O3 -DNDEBUG -DGTEST_HAS_PTHREAD=1 -MD -MT CMakeFiles/cartographer.mapping.3d.hybrid_grid_test.dir/cartographer/mapping/3d/hybrid_grid_test.cc.o -MF CMakeFiles/cartographer.mapping.3d.hybrid_grid_test.dir/cartographer/mapping/3d/hybrid_grid_test.cc.o.d -o CMakeFiles/cartographer.mapping.3d.hybrid_grid_test.dir/cartographer/mapping/3d/hybrid_grid_test.cc.o -c /home/gcong/catkin_ws/src/cartographer/cartographer/mapping/3d/hybrid_grid_test.cc
In file included from /usr/include/c++/9/map:61,
                 from /home/gcong/catkin_ws/install_isolated/include/ceres/ordered_groups.h:34,
                 from /home/gcong/catkin_ws/install_isolated/include/ceres/ceres.h:52,
                 from /home/gcong/catkin_ws/src/cartographer/cartographer/common/math.h:25,
                 from /home/gcong/catkin_ws/src/cartographer/cartographer/mapping/3d/hybrid_grid.h:28,
                 from /home/gcong/catkin_ws/src/cartographer/cartographer/mapping/3d/hybrid_grid_test.cc:17:

/usr/include/c++/9/bits/stl_map.h: In instantiation of ‘std::map<_Key, _Tp, _Compare, _Alloc>::map(_InputIterator, _InputIterator) [with _InputIterator = cartographer::mapping::DynamicGrid<cartographer::mapping::NestedGrid<cartographer::mapping::FlatGrid<short unsigned int, 3>, 3> >::Iterator; _Key = Eigen::Matrix<int, 3, 1>; _Tp = float; _Compare = cartographer::mapping::{anonymous}::EigenComparator; _Alloc = std::allocator<std::pair<const Eigen::Matrix<int, 3, 1>, float> >]’:
/home/gcong/catkin_ws/src/cartographer/cartographer/mapping/3d/hybrid_grid_test.cc:216:47:   required from here
/usr/include/c++/9/bits/stl_map.h:273:4: error: no matching function for call to ‘std::_Rb_tree<Eigen::Matrix<int, 3, 1>, std::pair<const Eigen::Matrix<int, 3, 1>, float>, std::_Select1st<std::pair<const Eigen::Matrix<int, 3, 1>, float> >, cartographer::mapping::{anonymous}::EigenComparator, std::allocator<std::pair<const Eigen::Matrix<int, 3, 1>, float> > >::_M_insert_range_unique(cartographer::mapping::DynamicGrid<cartographer::mapping::NestedGrid<cartographer::mapping::FlatGrid<short unsigned int, 3>, 3> >::Iterator&, cartographer::mapping::DynamicGrid<cartographer::mapping::NestedGrid<cartographer::mapping::FlatGrid<short unsigned int, 3>, 3> >::Iterator&)’
  273 |  { _M_t._M_insert_range_unique(__first, __last); }
      |    ^~~~
In file included from /usr/include/c++/9/map:60,
                 from /home/myname/catkin_ws/install_isolated/include/ceres/ordered_groups.h:34,
                 from /home/myname/catkin_ws/install_isolated/include/ceres/ceres.h:52,
                 from /home/myname/catkin_ws/src/cartographer/cartographer/common/math.h:25,
                 from /home/myname/catkin_ws/src/cartographer/cartographer/mapping/3d/hybrid_grid.h:28,
                 from /home/myname/catkin_ws/src/cartographer/cartographer/mapping/3d/hybrid_grid_test.cc:17:
/usr/include/c++/9/bits/stl_tree.h:1109:2: note: candidate: ‘template<class _InputIterator> std::__enable_if_t<std::is_same<_Val, typename std::iterator_traits<_InputIterator>::value_type>::value> std::_Rb_tree<_Key, _Val, _KeyOfValue, _Compare, _Alloc>::_M_insert_range_unique(_InputIterator, _InputIterator) [with _InputIterator = _InputIterator; _Key = Eigen::Matrix<int, 3, 1>; _Val = std::pair<const Eigen::Matrix<int, 3, 1>, float>; _KeyOfValue = std::_Select1st<std::pair<const Eigen::Matrix<int, 3, 1>, float> >; _Compare = cartographer::mapping::{anonymous}::EigenComparator; _Alloc = std::allocator<std::pair<const Eigen::Matrix<int, 3, 1>, float> >]’
 1109 |  _M_insert_range_unique(_InputIterator __first, _InputIterator __last)
      |  ^~~~~~~~~~~~~~~~~~~~~~
/usr/include/c++/9/bits/stl_tree.h:1109:2: note:   template argument deduction/substitution failed:
/usr/include/c++/9/bits/stl_tree.h: In substitution of ‘template<class _InputIterator> std::__enable_if_t<std::is_same<std::pair<const Eigen::Matrix<int, 3, 1>, float>, typename std::iterator_traits<_Iterator>::value_type>::value, void> std::_Rb_tree<Eigen::Matrix<int, 3, 1>, std::pair<const Eigen::Matrix<int, 3, 1>, float>, std::_Select1st<std::pair<const Eigen::Matrix<int, 3, 1>, float> >, cartographer::mapping::{anonymous}::EigenComparator, std::allocator<std::pair<const Eigen::Matrix<int, 3, 1>, float> > >::_M_insert_range_unique<_InputIterator>(_InputIterator, _InputIterator) [with _InputIterator = cartographer::mapping::DynamicGrid<cartographer::mapping::NestedGrid<cartographer::mapping::FlatGrid<short unsigned int, 3>, 3> >::Iterator]’:
/usr/include/c++/9/bits/stl_map.h:273:4:   required from ‘std::map<_Key, _Tp, _Compare, _Alloc>::map(_InputIterator, _InputIterator) [with _InputIterator = cartographer::mapping::DynamicGrid<cartographer::mapping::NestedGrid<cartographer::mapping::FlatGrid<short unsigned int, 3>, 3> >::Iterator; _Key = Eigen::Matrix<int, 3, 1>; _Tp = float; _Compare = cartographer::mapping::{anonymous}::EigenComparator; _Alloc = std::allocator<std::pair<const Eigen::Matrix<int, 3, 1>, float> >]’
/home/gcong/catkin_ws/src/cartographer/cartographer/mapping/3d/hybrid_grid_test.cc:216:47:   required from here
/usr/include/c++/9/bits/stl_tree.h:1109:2: error: no type named ‘value_type’ in ‘struct std::iterator_traits<cartographer::mapping::DynamicGrid<cartographer::mapping::NestedGrid<cartographer::mapping::FlatGrid<short unsigned int, 3>, 3> >::Iterator>’
/usr/include/c++/9/bits/stl_map.h: In instantiation of ‘std::map<_Key, _Tp, _Compare, _Alloc>::map(_InputIterator, _InputIterator) [with _InputIterator = cartographer::mapping::DynamicGrid<cartographer::mapping::NestedGrid<cartographer::mapping::FlatGrid<short unsigned int, 3>, 3> >::Iterator; _Key = Eigen::Matrix<int, 3, 1>; _Tp = float; _Compare = cartographer::mapping::{anonymous}::EigenComparator; _Alloc = std::allocator<std::pair<const Eigen::Matrix<int, 3, 1>, float> >]’:
/home/gcong/catkin_ws/src/cartographer/cartographer/mapping/3d/hybrid_grid_test.cc:216:47:   required from here
/usr/include/c++/9/bits/stl_tree.h:1118:2: note: candidate: ‘template<class _InputIterator> std::__enable_if_t<(! std::is_same<_Val, typename std::iterator_traits<_InputIterator>::value_type>::value)> std::_Rb_tree<_Key, _Val, _KeyOfValue, _Compare, _Alloc>::_M_insert_range_unique(_InputIterator, _InputIterator) [with _InputIterator = _InputIterator; _Key = Eigen::Matrix<int, 3, 1>; _Val = std::pair<const Eigen::Matrix<int, 3, 1>, float>; _KeyOfValue = std::_Select1st<std::pair<const Eigen::Matrix<int, 3, 1>, float> >; _Compare = cartographer::mapping::{anonymous}::EigenComparator; _Alloc = std::allocator<std::pair<const Eigen::Matrix<int, 3, 1>, float> >]’
 1118 |  _M_insert_range_unique(_InputIterator __first, _InputIterator __last)
      |  ^~~~~~~~~~~~~~~~~~~~~~
/usr/include/c++/9/bits/stl_tree.h:1118:2: note:   template argument deduction/substitution failed:
[... cut ...]
      |                ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
[33/278] Building CXX object CMakeFiles/cartographer.dir/cartographer/mapping/3d/submap_3d.cc.o
ninja: build stopped: subcommand failed.
<== Failed to process package 'cartographer': 
  Command '['/home/gcong/catkin_ws/install_isolated/env.sh', 'ninja', '-j16', '-l16']' returned non-zero exit status 1.

Reproduce this error by running:
==> cd /home/my_nbame/catkin_ws/build_isolated/cartographer && /home/gcong/catkin_ws/install_isolated/env.sh ninja -j16 -l16

Command failed, exiting.

ArtlyStyles, 2020-10-20:
I noticed this:

/usr/bin/c++  -DBOOST_ALL_NO_LIB -DBOOST_IOSTREAMS_DYN_LINK -DGFLAGS_IS_A_DLL=0 -I/home/gcong/catkin_ws/src/cartographer/cartographer -I. -I/home/gcong/catkin_ws/src/cartographer -isystem /usr/src/googletest/googlemock/gtest/include -isystem /usr/include/eigen3 -isystem /home/gcong/catkin_ws/install_isolated/include -isystem /usr/include/lua5.2 -isystem /usr/include/cairo -isystem /usr/include/glib-2.0 -isystem /usr/lib/x86_64-linux-gnu/glib-2.0/include -isystem /usr/include/pixman-1 -isystem /usr/include/uuid -isystem /usr/include/freetype2 -isystem /usr/include/libpng16 -isystem /usr/src/googletest/googlemock/include -isystem /usr/src/googletest/googlemock -isystem /usr/src/googletest/googletest/include -isystem /usr/src/googletest/googletest -O3 -DNDEBUG    -pthread -std=c++11 -fPIC  -Wall -Wpedantic -Werror=format-security -Werror=missing-braces -Werror=reorder -Werror=return-type -Werror=switch -Werror=uninitialized -O3 -DNDEBUG -pthread -std=c++11 -fPIC  -Wall -Wpedantic -Werror=format-security -Werror=missing-braces -Werror=reorder -Werror=return-type -Werror=switch -Werror=uninitialized -O3 -DNDEBUG -DGTEST_HAS_PTHREAD=1 -MD -MT CMakeFiles/cartographer.mapping.3d.hybrid_grid_test.dir/cartographer/mapping/3d/hybrid_grid_test.cc.o -MF CMakeFiles/cartographer.mapping.3d.hybrid_grid_test.dir/cartographer/mapping/3d/hybrid_grid_test.cc.o.d -o CMakeFiles/cartographer.mapping.3d.hybrid_grid_test.dir/cartographer/mapping/3d/hybrid_grid_test.cc.o -c /home/gcong/catkin_ws/src/cartographer/cartographer/mapping/3d/hybrid_grid_test.cc
In file included from /home/myname/catkin_ws/src/cartographer/cartographer/mapping/3d/hybrid_grid_test.cc:17:
/home/gcong/catkin_ws/src/cartographer/cartographer/mapping/3d/hybrid_grid.h:31:10: fatal error: cartographer/mapping/proto/3d/hybrid_grid.pb.h: No such file or directory
   31 | #include "cartographer/mapping/proto/3d/hybrid_grid.pb.h"
      |          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.

johuber, 2020-10-20:
did you build following the instructions [here](https://google-cartographer-ros.readthedocs.io/en/latest/compilation.html#building-installation) ?

johuber, 2020-10-21:
so @ArtlyStyles , looking at #1529  it seems like you managed to build it!?
I will close this issue!

closed 2020-10-21 by johuber

songxuekun, 2023-07-12:
@ArtlyStyles do you solve the problem