cartographer-project/cartographer_ros #1531: Build instructions are not up to date

MichaelGrupp, 2020-10-22:
I tried building with a fresh, clean installation of ROS Noetic and it's not great.

Fix the instructions such that they make sense and work.

---

Observations so far:

`install_proto3.sh`:
- not needed on Noetic & Melodic, comes with rosdep. Potentially harmful.

Abseil:
- not mentioned
- `install_abseil.sh` is provided by cartographer - installation to a catkin build dir would be nice

ludoux, 2020-10-23:
Thank you, I face a build problem on ROS Noetic by following the official guide.
`CMake Error: The following variables are used in this project, but they are set to NOTFOUND.`

By installing the Abseil (`install_abseil.sh`) and the following cmake file change:
>this is required: cartographer-project/cartographer#1705 (the fix in FindGmock)

It works. Thank you again.

MichaelGrupp, 2020-10-27:
done

closed 2020-10-27 by MichaelGrupp

W-E-A, 2023-03-12:
Run "sudo apt-get install libgmock-dev" works for me.

commit 70612a4d (2020-10-27): Fix outdated build instructions (closes #1531). (#1537)