cartographer-project/cartographer_ros #1534: Incorrect scan matching during a robot rotation

ywiyogo, 2020-10-24:
My lua-option is defined like following:

    map_frame = "map",
    tracking_frame = "base_link",
    published_frame = "base_link",
    odom_frame = "odom",
    provide_odom_frame = false,
    publish_frame_projected_to_2d = false,
    use_pose_extrapolator = true,
    use_odometry = false,

Currently, I just plug my RPlidar A2 to my PC and testing the map, wihtout odometry/IMU sensor. If I rotate the lidar scanner, I can see that the topic `/scan_matched_points2` is somehow drifted and distorts the map generation. Please see the picture below ( red is the laser scanner, green is the `scan_matched_point2`:

[image]

If I see the demo from the Deutsches Museum bagfile, I don't see such a drift. What did I do wrong? Since the Laser frame is the same as the robot frame (base_link). Is it because I don't have odometry sensor? If I change the lua setting to `odom_frame = "base_link"`, I see the same result.

Thanks!

johuber, 2020-10-26:
tuning is tricky and maybe you find a way to improve it if you play around long enough.

However, rotations are extremely hard to handle (for almost any SLAM) without odometry info or an IMU.  Adding either of these will help you deal with this much easier.

ywiyogo, 2020-10-26:
> tuning is tricky and maybe you find a way to improve it if you play around long enough.
> 
> However, rotations are extremely hard to handle (for almost any SLAM) without odometry info or an IMU. Adding either of these will help you deal with this much easier.

Thank @johuber for the hint. I'll get an IMU sensor then.