cartographer-project/cartographer_ros #1536: New ROS 2 port of cartographer_ros - tracking issue

MichaelGrupp, 2020-10-26:
We plan to make a new ROS 2 port of cartographer_ros that is in sync with master.

There is a port of an older version that can be used as an inspiration for a new port: https://github.com/cartographer-project/cartographer_ros/pull/1488

This issue aims to track progress and provide an entry point for people who are interested to contribute.

(_work in progress_)

3473f, 2020-10-26:
Would definitely love to get involved as I have been getting my hands dirty with the source code recently. Any leads on where to start would be appreciated.

Serafadam, 2020-10-29:
Hi, I'd also like to help, for starters, I am currently using Cartographer in one of my projects, from the ros2 repository. If I want to move to ROS2 Foxy, I'll also need to upgrade rplidar repository to this distro. As for creating this version, will there be some kind of release? I'd rather not start working before some huge feature gets added. As for the way of working, I think the best would be to branch off the latest commit here, and base work on https://github.com/cartographer-project/cartographer_ros/pull/1488.

How does that sound?

MichaelGrupp, 2020-10-30:
Hi @3473f and @Serafadam, thanks for joining the discussion! I have just updated the description with more details that we discussed this week in a maintainers meeting. Would be great if you can have a look and let us know what you think and if this is something you would like to contribute to.

---

> As for creating this version, will there be some kind of release?

Yes, that's the goal. To have a new release and then in the future also more frequent releases.

> As for the way of working, I think the best would be to branch off the latest commit here, and base work on #1488.

I also think so, as I wrote now in the description.

3473f, 2020-11-07:
Hi @MichaelGrupp, sorry for the late reply! that definitely sounds interesting. I have a lot of my plate in the next two weeks, after which I can start contributing regularly to this port.

> Create a new branch from master and apply similar changes like the ones that were added in the original ROS2 port.
I'd rather follow this approach as well. The only case I'd find starting from the current ROS2 port useful is if it was a few commits behind, but it seems that Cartographer has changed a lot since then.

airballking, 2020-11-11:
@MichaelGrupp Great initiative! My team has good experience using Cartographer and wants to switch our current robot system from ROS1 to ROS2. Right now, Cartographer is the only dependency that is still blocking this migration. So, we are willing to help. Let us know how we can contribute!

MichaelGrupp, 2020-11-11:
Hi, to give you an update: I actually started doing some things myself already:

* we have a `ros2-dashing` branch that we can use for PRs
* I opened a PR for setting up a first continuous integration infrastructure: #1543 
* another PR for porting `cartographer_ros_msgs`: #1542

I started it this way because this repo actually includes 3 ROS packages `cartographer_{ros, ros_msgs, rviz}` and the message package is needed for the others (and also rather easy to port).

Once these are merged (I plan to follow up in the next days), we can go on with porting the `cartographer_ros` package, which is the biggest one. Here it would be awesome if someone of you could do the deep-dive into to the code, I probably won't have so much time for this in the next weeks.

What do you think? Feel free to comment also on the open current pull requests of course.

airballking, 2020-11-11:
@MichaelGrupp Sounds like a reasonable plan. Regarding deep-dive into ```cartographer_ros```, do you already have an overview which nodes/scripts can be ported on their own? Those could end-up being individual items in a check-list and separate PRs. With such a list, it could be easier to commit to taking on individual items.

MichaelGrupp, 2020-11-11:
@airballking I like this idea. I opened an extra issue for this because it maybe goes a bit too much into detail for this tracking issue: https://github.com/cartographer-project/cartographer_ros/issues/1557

handmat78, 2021-02-16:
If you need IMU data ROS2 driver, SBG Systems has developed one: available on the [repo](https://github.com/SBG-Systems/sbg_ros2)

doisyg, 2021-05-05:
>     * Create a new branch from `master` and apply similar changes like the ones that were added in the original ROS2 port.

This is the strategy that I applied here: https://github.com/cartographer-project/cartographer_ros/pull/1622
I believe most of the work is done. Still needs extensive testing, parameter check, performance review and complementary features to be converted (rviz!)

doisyg, 2021-11-18:
Rviz is now supported in ros2 galactic in our PR https://github.com/cartographer-project/cartographer_ros/pull/1622 . Any suggestions on how to move forward with a proper ros2 galactic release ? Current ros2 bin release is from a very old version of both the library and the wrapper (with only cartographer_node working).
We are planning to discuss and add lifecycle support next.

iLanceS, 2022-09-19:
Is there a planning for Port it to Humble Hawksbill ?
@doisyg

doisyg, 2022-09-19:
Yes, it was released to Rolling and Humble a couple of months ago already. See 
https://discourse.ros.org/t/rolling-and-soon-humble-release-of-both-cartographer-and-cartographer-ros-v2-and-call-for-testing/25137
and
https://www.linkedin.com/posts/guillaume-doisy-b707362b_super-happy-to-finally-see-cartographer-fully-activity-6924372671514247169-UK7L?utm_source=share&utm_medium=member_desktop