cartographer-project/cartographer_ros #1544: Problems building the map

francescoricciuti, 2020-11-02:
Hi all,

I'm having problems building a map.

I'm using a 31 min long bag, covering a road around 2.5 km long. I'm using the same configuration parameters I've used many other times with the same setup sensor+robot. I have a 16 layer lidar publishing at 20hz and an IMU publishing at 100HZ
Here the launch file and lua:
[launch.txt](https://github.com/cartographer-project/cartographer_ros/files/5476423/launch.txt)
[robot_3d.txt](https://github.com/cartographer-project/cartographer_ros/files/5476424/robot_3d.txt)

I'm using the "offline" node here, but the same happens playing the bag separately and running the "online" node.

I validated the bag with rosbag_validate and no error occur. Moreover, I've recorded two bag in two different days covering the same area, both giving the same error.

Here's what happens:

At first, it seems that everything goes as expected: the process goes on without any problem and the map looks great from Rviz.
Problems start when the bag ends, and the node says the following 

[image]

It looks like there's no trajectory 0, but I can see it from Rviz.
The process then goes on optimizing and finishes cleanly.

When I try to create a png+yaml using the cartographer_pbstream_to_ros_map node, it crashes with the following output

[image]

It seems that someone already had the same problem [here](https://github.com/cartographer-project/cartographer_ros/issues/1203), but it seems that there isn't a consensus on why it happened or on how to solve it.

Moreover, when I try to visualize the pbstream with the visualize_pbstream.launch file, here's the output

[image]

as if there's something wrong with trajectories and submaps. The strange thing is that I see the submaps being built and fading properly on Rviz while building the map.

Anyone has an idea on what it could be? Is it a tuning problem, a problem with the data I have or something that I'm doing wrong?

Thanks in advance,

Francesco

johuber, 2020-11-02:
your PC running out of RAM during the process? first thing I could think of..

francescoricciuti, 2020-11-02:
No, that's what I also thought of but I monitored the whole process with htop and RAM was heavily used, but never run out of it. I even tried to execute everything without Rviz open to avoid such problems.

francescoricciuti, 2020-11-06:
@MichaelGrupp , @johuber any idea on this? 

Here's the [bag](https://www.dropbox.com/s/ffdkuym9ifqheq8/bag_not_working.bag?dl=0) in case someone wants to help me with this!

Thanks!!

johuber, 2020-11-06:
> @MichaelGrupp , @johuber any idea on this?
> 
> Here's the [bag](https://www.dropbox.com/s/ffdkuym9ifqheq8/bag_not_working.bag?dl=0) in case someone wants to help me with this!
> 
> Thanks!!

I can't promise to look into this anytime soon, sorry. 

my next step would be to look at the data (check if timestamps are correct, check if there are any longer timeouts of any sensor that could cause issues, check if RVIZ can visualize them or if there is a warning..).

What sensor drivers are you using?

The setup seems pretty standard and the bag is also not particularly long, we've worked with much longer ones.

francescoricciuti, 2020-11-06:
> I can't promise to look into this anytime soon, sorry.

no problem, I appreciate your help so far!

I'm using a Robosense Lidar with its drivers and a custom driver for the IMU, but this is the same setup we have always used. It's surprising because we have worked with bags and mapped areas that are way bigger than this, but this is the first time I see this kind of error.

Regarding Rviz, it gives errors with constraints and trajectories, but it shows both of them anyway.

My hypothesis on this is that something wrong happens with the tf map->base_link, but it's just a feeling not supported by clear evidence unfortunately

closed 2020-12-01 by francescoricciuti