cartographer-project/cartographer_ros #1548: global SLAM makes the map drift more, why??

bxc237, 2020-11-05:
To improve the chances that other Cartographer users can help you,
here are some guidelines for describing your tuning problem:

1. check the tuning guide on the documentation page and previous issues -
   some problems might have been solved already by other people.

2. run `rosbag_validate` which does some checks on your sensor data. This
   tool often finds issues that can explain poor performance and must be fixed
   at recording time. Please post the full output of the tool into a
   GitHub Gist at https://gist.github.com/, then link it in the issue even if
   it does not report anything. You can run the tool like this:

   rosrun cartographer_ros cartographer_rosbag_validate -bag_filename <bag filename>

3. post a link to a Git repository containing a branch of
   `cartographer_ros` containing all the configuration, launch, and URDF files
   required to reproduce your issue.
4. post a link to a bag file we can use to reproduce your issue. Put it on
   Google Drive, Dropbox, any webserver or wherever it is publicly
   downloadable.
5. remove this boilerplate text before submitting your issue.

PLEASE NOTE: tuning issues are not necessarily handled by the maintainers and
can be closed after a period of inactivity.

soorajanilkumar, 2020-11-24:
Hi @bxc237 

I tried your rosbag. There seems to be some kind of corruption at the end. If I filter out the last few seconds and use the config below , I get a good map with loop closure.

I used your config as a base and filtered out some parameters which I though would improve matching.

Granted I tested the rosbag using the offline mode, so you might have to tune for latency if you want online slam performance.

```
include "map_builder.lua"
include "trajectory_builder.lua"

options = {
  map_builder = MAP_BUILDER,
  trajectory_builder = TRAJECTORY_BUILDER,
  map_frame = "map_2D_test",
  tracking_frame = "base_footprint",
  published_frame = "odom_2D",
  odom_frame = "odom_2D",
  provide_odom_frame = false,
  publish_frame_projected_to_2d = false,
  use_odometry = true,
  use_nav_sat = false,
  use_landmarks = false,
  num_laser_scans = 1,
  num_multi_echo_laser_scans = 0,
  num_subdivisions_per_laser_scan = 1,
  num_point_clouds = 0,
  lookup_transform_timeout_sec = 0.2,
  submap_publish_period_sec = 0.5,
  pose_publish_period_sec = 0.5,
  trajectory_publish_period_sec = 0.5,
  rangefinder_sampling_ratio = 1.,
  odometry_sampling_ratio = 1.,
  fixed_frame_pose_sampling_ratio = 1.,
  imu_sampling_ratio = 1.,
  landmarks_sampling_ratio = 1.,
}

MAP_BUILDER.use_trajectory_builder_2d = true

TRAJECTORY_BUILDER_2D.min_range = 0.1
TRAJECTORY_BUILDER_2D.max_range = 12.
TRAJECTORY_BUILDER_2D.missing_data_ray_length = 12.
TRAJECTORY_BUILDER_2D.use_imu_data = false
TRAJECTORY_BUILDER_2D.use_online_correlative_scan_matching = true

POSE_GRAPH.optimize_every_n_nodes = 20 -- for offline mapping (might be different for online case)
TRAJECTORY_BUILDER_2D.submaps.num_range_data = 80

TRAJECTORY_BUILDER_2D.motion_filter.max_angle_radians = math.rad(1) -- 1 degree seems more apt
TRAJECTORY_BUILDER_2D.motion_filter.max_distance_meters = 0.1

POSE_GRAPH.constraint_builder.min_score = 0.6 -- increased threshold to throw away wrong matches
POSE_GRAPH.constraint_builder.global_localization_min_score = 0.4

POSE_GRAPH.constraint_builder.max_constraint_distance = 20.

-- odometry was good, so i used for atleast translation to weed out any incorrect constraints
POSE_GRAPH.optimization_problem.odometry_translation_weight = 1000
POSE_GRAPH.optimization_problem.odometry_rotation_weight = 10

-- increased search radius
POSE_GRAPH.constraint_builder.fast_correlative_scan_matcher.linear_search_window = 15

return options

```

Main changes
- increased search radius
- increased motion filter from 0.1 degrees to 1 degree for latency
- increased min score to weed out incorrect matches

I am by no means an expert and there might be other things which can be improved. @MichaelGrupp  can advice on further modifications. 

Cheers

bxc237, 2021-05-07:
Hi @soorajanilkumar ,
Thank you for your awesome advice! Soon after I posted this issue, we found the solution so that I forgot to close this issue. You are right, the key points are "increase min score to weed out incorrect matches" and "increase POSE_GRAPH.optimize_every_n_nodes to 20~30".

closed 2021-05-07 by bxc237