cartographer-project/cartographer_ros #1550: "Encountered a point with a larger stamp than the last point in the cloud"

Oburshan, 2020-11-05:
Hey, i am getting this fatal error when launching the pure localization mode live, with VLP-16.
Its happen sometimes and shutting down the node.

[FATAL] [1604575697.682604927]: F1105 13:28:17.000000 10916 msg_conversion.cc:273] Check failed: point.time <= 0.f (1.27498e-06 vs. 0) Encountered a point with a larger stamp than the last point in the cloud.

Thanks!

Nova555, 2020-12-29:
Have you solved the problem?why?

Oburshan, 2020-12-29:
I “Filtered” the frames that contained this problem and it is work great.
I don’t understand why the original code kills the node on this scenario when they could just throw this specific pointcloud frame sample.

Nova555, 2020-12-29:
Redownload and catkin_make

Oburshan, 2020-12-29:
Thank's but I've already fixed it, I think there is a deeper problem that whenever one sensor sample does not meet a particular requirement it kills the node instead of throwing away this specific sample.
I will try to fix it in the original code and pull request it.

Nova555, 2021-05-06:
Hi，this problem reappeared. Can you share your code to fix this problem?

hengjiUSTC, 2021-06-14:
We'v met same problem, is there any option that we can use to let cartographer ignore the topic instead of throw crashing?

nonlinear1, 2021-08-04:
Me, too. I also get this fatal error! Could some know how to solve it for now?

nonlinear1, 2021-08-04:
@Oburshan Do you have any way to solve this problem?

patcmorneau, 2022-11-22:
> I filtered the frames that contained this problem and it is working great. I don’t understand why the original code kills the node on this scenario when they could just throw this specific pointcloud frame sample.

how did you filtered the problematic message ?

nonlinear1, 2022-11-23:
@patcmorneau the method which is given by Nova555 👍 Modify " {Eigen::Vector3f{point.x, point.y, point.z}, point.time}); " to " {Eigen::Vector3f{point.x, point.y, point.z}, 0.f}); " with msg_conversion.cc is work for me.

nonlinear1, 2022-11-23:
Although cartograher is very good at code style and software architecture. for this timestamp processing, it is bad

patcmorneau, 2022-11-23:
removed all topics but imu and lidar from my bag and the error is gone now ... will see if i can generate a map now