cartographer-project/cartographer_ros #1552: [pure localization ERROR] when i run the demo_backpack_2d_localization.launch

wesigj, 2020-11-07:
In the actual operation of the robot, with IMU and ODOM, it is normally running at the beginning, but after running for a period of time, the following errors will occur, resulting in the failure of normal operation. I hope you can help me, thank you very much

*** Check failure stack trace: ***
    @     0x7f81b84a05cd  google::LogMessage::Fail()
    @     0x7f81b84a2433  google::LogMessage::SendToLog()
    @     0x7f81b84a015b  google::LogMessage::Flush()
    @     0x7f81b84a2e1e  google::LogMessageFatal::~LogMessageFatal()
    @           0x6ae916  cartographer::mapping::optimization::OptimizationProblem2D::AddOdometryData()
    @           0x6c6bd1  _ZNSt17_Function_handlerIFN12cartographer7mapping8WorkItem6ResultEvEZNS1_11PoseGraph2D15AddOdometryDataEiRKNS0_6sensor12OdometryDataEEUlvE_E9_M_invokeERKSt9_Any_data
    @           0x6ca884  cartographer::mapping::PoseGraph2D::DrainWorkQueue()
    @           0x6d367b  cartographer::mapping::PoseGraph2D::HandleWorkQueue()
    @           0x71f9f2  cartographer::mapping::constraints::ConstraintBuilder2D::RunWhenDoneCallback()
    @           0x678ab1  cartographer::common::Task::Execute()
    @           0x676959  cartographer::common::ThreadPool::DoWork()
    @     0x7f81b6293c80  (unknown)
    @     0x7f81b7b186ba  start_thread
    @     0x7f81b59f94dd  clone
    @              (nil)  (unknown)
[ WARN] [1604721104.902795190]: Costmap2DROS transform timeout. Current time: 1604721104.9026, global_pose stamp: 1604721103.3629, tolerance: 0.5000
[ WARN] [1604721104.903309763]: Could not get robot pose, cancelling reconfiguration
[cartographer_node-10] process has died [pid 8074, exit code -6, cmd /home/lyh/catkin_carto/install_isolated/lib/cartographer_ros/cartographer_node -configuration_directory /home/lyh/catkin_carto/install_isolated/share/cartographer_ros/configuration_files -configuration_basename backpack_2d_localization.lua -load_state_filename /home/lyh/lyric_ws/src/Core/lyric_maps_manager/scripts/map/lyric_05.pbstream echoes:=scan imu:=imu_data __name:=cartographer_node __log:=/home/lyh/.ros/log/6458be20-20ac-11eb-9ce9-70e04f682b80/cartographer_node-10.log].
log file: /home/lyh/.ros/log/6458be20-20ac-11eb-9ce9-70e04f682b80/cartographer_node-10*.log
[ WARN] [1604721105.903731810]: Costmap2DROS transform timeout. Current time: 1604721105.9035, global_pose stamp: 1604721103.3629, tolerance: 0.5000
[ WARN] [1604721105.904021126]: Could not get robot pose, cancelling reconfiguration
[ WARN] [1604721106.940985045]: Costmap2DROS transform timeout. Current time: 1604721106.9408, global_pose stamp: 1604721103.3629, tolerance: 0.5000
[ WARN] [1604721106.968826656]: Could not get robot pose, cancelling reconfiguration
[image]
[image]

```

  map_frame = "map",
  tracking_frame = "laser_frame",
  published_frame = "odom_combined",
  odom_frame = "odom_combined",
  provide_odom_frame = false,
  publish_frame_projected_to_2d = false,
  use_pose_extrapolator = true,
  use_odometry = true,
  use_nav_sat = false,
  use_landmarks = false,
  num_laser_scans = 1,
  num_multi_echo_laser_scans = 0,
  num_subdivisions_per_laser_scan = 1,
  num_point_clouds = 0,
  lookup_transform_timeout_sec = 0.2,
  submap_publish_period_sec = 0.3,
  pose_publish_period_sec = 5e-3,
  trajectory_publish_period_sec = 30e-3,
  rangefinder_sampling_ratio = 1.,
  odometry_sampling_ratio = 1.,
  fixed_frame_pose_sampling_ratio = 1.,
  imu_sampling_ratio = 1.,
  landmarks_sampling_ratio = 1.,
}

MAP_BUILDER.use_trajectory_builder_2d = true

TRAJECTORY_BUILDER_2D.submaps.num_range_data = 35
TRAJECTORY_BUILDER_2D.min_range = 0.3
TRAJECTORY_BUILDER_2D.max_range = 8.
TRAJECTORY_BUILDER_2D.missing_data_ray_length = 1.
TRAJECTORY_BUILDER_2D.use_imu_data = false
TRAJECTORY_BUILDER_2D.use_online_correlative_scan_matching = true
TRAJECTORY_BUILDER_2D.real_time_correlative_scan_matcher.linear_search_window = 0.1
TRAJECTORY_BUILDER_2D.real_time_correlative_scan_matcher.translation_delta_cost_weight = 10.
TRAJECTORY_BUILDER_2D.real_time_correlative_scan_matcher.rotation_delta_cost_weight = 1e-1

POSE_GRAPH.optimization_problem.huber_scale = 1e2
POSE_GRAPH.optimize_every_n_nodes = 35
POSE_GRAPH.constraint_builder.min_score = 0.65
```

MichaelGrupp, 2020-11-23:
A `CHECK()` statement in the code failed. Normally it prints the details of the check in the log before the stacktrace, somehow this seems to be missing in your log.

But since it happens in the `MapByTime::Append()` call, it's probably this check here that fails: https://github.com/cartographer-project/cartographer/blob/fdcf0eb0df72b2eb7ae30c08c873b5ab2b855239/cartographer/sensor/map_by_time.h#L43

This happens when the timestamps of the sensor data are not strictly increasing. Cartographer is very picky here, so you need to ensure that your odometry, IMU and scan topics have correct timestamps.

wesigj, 2020-11-24:
Thank you so much for your assistance. i had solve this problem for using my correct topic of odom_tf ,because it is publish by robot_pose_ekf

closed 2020-11-25 by wesigj