cartographer-project/cartographer_ros #1557: ROS2 port of cartographer_ros (library & executables)

MichaelGrupp, 2020-11-11:
## Components

* library (see `cartographer_ros/CMakeLists.txt`)
* nodes & other executables (see `cartographer_ros/cartographer_ros/CMakeLists.txt`)
* the package.xml (and other relevant files for compiling in CI) would need to be adapted too of course

The library cmake file uses a glob filter to get all `.cc` files. This could be changed for the period of porting, to allow porting only a subset of files at a time (in correct dependency order of course).

Regarding executables I would say the first goal should be to port `node_main.cc`, then `offline_node_main.cc` and then all the remaining tools. Files that don't build yet should be excluded from the build temporarily, by commenting out in the cmake file.

## How to test locally with Docker

This is how we would test before merging, so it makes also sense to test it like this locally.

Once #1543 is merged, you can activate the build of cartographer_ros in `Dockerfile.foxy` and in `scripts/install_debs.sh` (see TODO comments in these files). Then run
```
docker build . --file Dockerfile.foxy --tag cartographer_ros:foxy
```

jediofgever, 2020-12-18:
Hey, 
just wanted to know whether ros2 branch builds with foxy ?. Please let me know if anyone has tried. Thank you

seasony-vp, 2020-12-18:
> Hey,
> just wanted to know whether ros2 branch builds with foxy ?. Please let me know if anyone has tried. Thank you

It seems that the install fails. I created an issue #1579.

MichaelGrupp, 2021-01-11:
@jediofgever Work in progress. Not yet ready to build.

@seasony-vp This is unrelated to this issue. You are referring to the old port made by OSRF, we plan to make a new one here.

imstevenpmwyca, 2021-02-12:
Dear @MichaelGrupp 

> Work in progress. Not yet ready to build.

Do you have any updates regarding this?

MichaelGrupp, 2021-03-15:
I didn't have the time for continuing the WIP state recently. But I will update here once there's something new.

doisyg, 2021-05-05:
PR for rolling/galactic here https://github.com/cartographer-project/cartographer_ros/pull/1622
(should maybe target a new branch ?)