cartographer-project/cartographer_ros #1560: Cartographer output cannot allign frames.

trevgilk, 2020-11-14:
Hi,

We tried cartographer with a handheld single velodyne 32c and an IMU. Unless there's very little movement or the lidar is static the slam is very messy and appears to be a disorganized “blob,” as can be seen in the attached image. We think there could be an issue in the .urdf file and We're trying to update the values to reflect our setup. We are wondering if anything in the .launch file or .lua file isn't setup properly.

[Bag file](https://drive.google.com/file/d/1vYYBhSPaNGl6L2gOfNxGHSHDm-e4FCAz/view)

[.urdf file](https://github.com/trevgilk/cartographer_ros/blob/master/cartographer_ros/urdf/ingenium_lidar_stick.urdf)

[slam.launch file](https://github.com/trevgilk/cartographer_ros/blob/master/cartographer_ros/launch/ingenium_slam.launch)

[slam.lua file](https://github.com/trevgilk/cartographer_ros/blob/master/cartographer_ros/configuration_files/ingenium_slam.lua)

These are the files we used to generate the point cloud and images:
[localization.launch file](https://github.com/trevgilk/cartographer_ros/blob/master/cartographer_ros/launch/ingenium_localization.launch)

[localization.lua](https://github.com/trevgilk/cartographer_ros/blob/master/cartographer_ros/configuration_files/ingenium_localization.lua)

[Here ](https://github.com/JohannesByle/ingenium_cartographer)is the repository we used to work on the .launch files and .urdf files

Pictures:
Static:
[image]

With movement:
[image]

eithwa, 2021-08-19:
[image]
I modified urdf and lua to make it work slightly
[ingenium.zip](https://github.com/cartographer-project/cartographer_ros/files/7011577/ingenium.zip)

JohannesByle, 2021-08-23:
Thank you for answering! We ended up fixing the problem a while ago, the problem ended up being with the urdf, we had the imu pointed upside down and the lidar not rotated properly. @trevgilk I suggest closing this issue.

closed 2021-08-23 by trevgilk