cartographer-project/cartographer_ros #1566: IMU calibration error clarification

soorajanilkumar, 2020-11-20:
Hi,

When I turned ON log_solver_summary during offline 3D mapping, I found out that there was about 179 degree of IMU correction. This surprised me, as I don't think I made a mistake in my TFs. 

When I tried to correct my TF's according to imu calibration #1025 , mapping went haywire. 
Can anyone please help me understand what causes the imu calibration error of this magnitude? 

Log provided below:
```
                                     Original                  Reduced
Parameter blocks                        55790                     2832
Parameters                             195263                     9910
Effective parameters                   167367                     8494
Residual blocks                          5477                     5477
Residual                                24429                    24429

Minimizer                        TRUST_REGION

Sparse linear algebra library    SUITE_SPARSE
Trust region strategy     LEVENBERG_MARQUARDT

                                        Given                     Used
Linear solver          SPARSE_NORMAL_CHOLESKY   SPARSE_NORMAL_CHOLESKY
Threads                                     7                        7
Linear solver threads                       1                        1
Linear solver ordering              AUTOMATIC                     2832

Cost:
Initial                          8.754483e+07
Final                            2.980566e+06
Change                           8.456427e+07

Minimizer iterations                        6
Successful steps                            6
Unsuccessful steps                          0

Time (in seconds):
Preprocessor                           0.0055

  Residual evaluation                  0.0027
  Jacobian evaluation                  0.0327
  Linear solver                        0.0791
Minimizer                              0.1224

Postprocessor                          0.0014
Total                                  0.1294

Termination:                      CONVERGENCE (Function tolerance reached. |cost_change|/cost: 9.634803e-07 <= 1.000000e-06)
[ INFO] [1605876369.520903889, 1605106911.719728715]: I1120 13:46:09.000000  3815 optimization_problem_3d.cc:573] Gravity was: 9.76214
[ INFO] [1605876369.520920701, 1605106911.726605078]: I1120 13:46:09.000000  3815 optimization_problem_3d.cc:575] IMU correction was: 179.447 deg (-0.00482998, 0.0277695, 0.000119932, -0.999603)
[ INFO] [1605876369.520933585, 1605106911.726605078]: I1120 13:46:09.000000  3815 optimization_problem_3d.cc:571] Trajectory 1:
[ INFO] [1605876369.520945347, 1605106911.726605078]: I1120 13:46:09.000000  3815 optimization_problem_3d.cc:573] Gravity was: 9.75924
[ INFO] [1605876369.520959875, 1605106911.726605078]: I1120 13:46:09.000000  3815 optimization_problem_3d.cc:575] IMU correction was: 179.604 deg (0.00345511, -0.0282802, 0.00193618, 0.999592)
[ INFO] [1605876369.520973450, 1605106911.726605078]: I1120 13:46:09.000000  3815 optimization_problem_3d.cc:571] Trajectory 2:
[ INFO] [1605876369.520988028, 1605106911.729587146]: I1120 13:46:09.000000  3815 optimization_problem_3d.cc:573] Gravity was: 9.74689
[ INFO] [1605876369.521005771, 1605106911.729587146]: I1120 13:46:09.000000  3815 optimization_problem_3d.cc:575] IMU correction was: 179.379 deg (-0.00541816, 0.0238153, -0.000648379, -0.999701)
[ INFO] [1605876369.540338817, 1605106911.769775671]: I1120 13:46:09.000000  3815 pose_graph_3d.cc:848] Translational residuals histogram:
Count: 2614  Min: 0.00201621  Max: 0.164882  Mean: 0.0320144
[0.002016, 0.018303)	               #####	Count: 637 (24.3688%)	Total: 637 (24.3688%)
[0.018303, 0.034589)	           #########	Count: 1120 (42.8462%)	Total: 1757 (67.215%)
[0.034589, 0.050876)	                ####	Count: 481 (18.4009%)	Total: 2238 (85.6159%)
[0.050876, 0.067162)	                  ##	Count: 204 (7.80413%)	Total: 2442 (93.42%)
[0.067162, 0.083449)	                   #	Count: 95 (3.63428%)	Total: 2537 (97.0543%)
[0.083449, 0.099735)	                    	Count: 51 (1.95103%)	Total: 2588 (99.0054%)
[0.099735, 0.116022)	                    	Count: 16 (0.612089%)	Total: 2604 (99.6174%)
[0.116022, 0.132308)	                    	Count: 4 (0.153022%)	Total: 2608 (99.7705%)
[0.132308, 0.148595)	                    	Count: 4 (0.153022%)	Total: 2612 (99.9235%)
[0.148595, 0.164882]	                    	Count: 2 (0.0765111%)	Total: 2614 (100%)
[ INFO] [1605876369.540407466, 1605106911.769775671]: I1120 13:46:09.000000  3815 pose_graph_3d.cc:850] Rotational residuals histogram:
Count: 2614  Min: 0.000515685  Max: 0.0651905  Mean: 0.00671999
[0.000516, 0.006983)	        ############	Count: 1604 (61.3619%)	Total: 1604 (61.3619%)
[0.006983, 0.013451)	             #######	Count: 883 (33.7796%)	Total: 2487 (95.1415%)
[0.013451, 0.019918)	                   #	Count: 92 (3.51951%)	Total: 2579 (98.6611%)
[0.019918, 0.026386)	                    	Count: 11 (0.420811%)	Total: 2590 (99.0819%)
[0.026386, 0.032853)	                    	Count: 3 (0.114767%)	Total: 2593 (99.1966%)
[0.032853, 0.039321)	                    	Count: 2 (0.0765111%)	Total: 2595 (99.2731%)
[0.039321, 0.045788)	                    	Count: 16 (0.612089%)	Total: 2611 (99.8852%)
[0.045788, 0.052256)	                    	Count: 2 (0.0765111%)	Total: 2613 (99.9617%)
[0.052256, 0.058723)	                    	Count: 0 (0%)	Total: 2613 (99.9617%)
[0.058723, 0.065190]	                    	Count: 1 (0.0382555%)	Total: 2614 (100%)

```

Thanks
Sooraj

johuber, 2020-11-20:
since the error is almost exactly 180degrees I am almost certain that the coordinate system of the IMU is different than you think. Some IMUs actually also do use slightly unexpected ways to define their coordinate frame, I have also struggled with this a bit in the past.
Also make sure that your IMU ROS-driver does not mess with the coordinate system or apply any transormations before publishing.

soorajanilkumar, 2020-11-20:
Hi @johuber ,

Thank you for your quick response. 

I had already corrected the IMU default convention from left handed (who uses that!) to std standard right-handed convention. But still I had the imu calibration error. 

I don't know why. For now, I re-fixed the IMU in the same orientation as base_link on the robot and currently the imu calibration error is now 5 degrees.
I guess there might be an error I missed somehow previously in the TFs.

Closing the issue for now. 
Thanks a lot once again.

closed 2020-11-20 by soorajanilkumar