cartographer-project/cartographer_ros #1567: Cartographer_ros build failed

harisiva1, 2020-11-23:
Hi,
I am trying to build the cartographer package using the below command line.
catkin_make_isolated --install --use-ninja
cartographer and cartographer_ros_msgs gets compiled but the **cartographer_ros** package gets failed and the reason is not clear for me.can anyone support?
[image]

johuber, 2020-11-24:
were the dependencies properly installed?
do you have `ros-melodic-geometry-msgs` ? It seems like it is trying to link it to eloquent stuff..

harisiva1, 2020-11-26:
sorry for the late reply.
regarding the dependencies --> yes everything is perfect.please see the below image
[image]

I am using both ros melodic and eloquent as i need both to work on a project.
regarding the geometry-msg i don't get what do you mean.how do i check it?
Thanks in advance!

johuber, 2020-11-30:
hmm okay so seems like you did that part correctly. Just to be sure cartographer does not have a missing dependency:
try `sudo apt install ros-melodic-geometry-msgs` and see if anything new is installed or not