cartographer-project/cartographer_ros #1577: cartographer offline mode with single threaded run

Howonjung, 2020-12-15:
Hello,

I am running a cartographer with offline mode ( [cartographer online and offline mode](https://github.com/cartographer-project/cartographer_ros/issues/281#issuecomment-288327633) ) 

so, basically I give a bag file and other configuration file as an input of cartographer offline mode.

operation itself seems not to have any problem but the problem is that

It is using only a single thread like picture below ( Only a CPU4 is maximally running, and the other is not operating intensively like CPU4 )

Is there any suggestion that I can fully utilize the other CPUs?

[image]

AchmadFathoni, 2021-02-12:
@Howonjung do you find any solution for this?

Howonjung, 2021-02-22:
@AchmadFathoni  I just downloaded the latest cartographer version and built. and it worked!

closed 2021-02-22 by Howonjung