cartographer-project/cartographer_ros #1579: Ros2 Foxy Installation Fails

seasony-vp, 2020-12-18:
Installing cartographer for [ros2 foxy](https://index.ros.org/p/cartographer/github-ros2-cartographer/#foxy) does not work.

The failure happens at this step from the [docs](https://google-cartographer-ros.readthedocs.io/en/latest/compilation.html#building-installation)
`rosdep install --from-paths src --ignore-src --rosdistro=${ROS_DISTRO} -y
`

The error message:
```
ERROR: the following packages/stacks could not have their rosdep keys resolved
to system dependencies:
cartographer_rviz: Cannot locate rosdep definition for [rviz]
cartographer_ros_msgs: Cannot locate rosdep definition for [message_runtime]
cartographer_ros: Cannot locate rosdep definition for [rosunit]
```
What's wrong? My hunch is that the [rosunit](https://index.ros.org/p/rosunit/), [message_runtime](https://index.ros.org/p/message_runtime/) and [rviz](https://index.ros.org/r/rviz/#foxy) packages haven't been released for foxy and "rviz" has now been updated  to "rviz2".

What are the appropriate instructions to install for foxy? Thanks.

seasony-vp, 2020-12-22:
Solved. To install with foxy use the downstream repo at ros2/cartographer.

Install by running this command: `sudo apt-get install ros-foxy-cartographer` as referenced [here](https://github.com/ros2/cartographer/issues/19#issuecomment-654301213).

closed 2020-12-22 by seasony-vp

MichaelGrupp, 2021-03-15:
The branch is currently in development. Sorry for the inconvenience.