cartographer-project/cartographer_ros #1580: Check failed: odometry_data.pose.IsValid()

TouchDeeper, 2020-12-28:
Hi,

When I want to run cartographer in gazebo, error occur:
```
global_trajectory_builder.cc:100] Check failed: odometry_data.pose.IsValid() { t: [inf, -inf, -0.000397655], q: [6.2123e-09, 3.52316e-08, -0.106439, -7.50879e-08] }
```

Be inspired by this [issue](https://github.com/cartographer-project/cartographer/issues/1110#issuecomment-384228458), I check my `/odom`, the pose msg is something like below, the value seems like no problem, the norm of the quaternion is 1. Could you please provide some advice? Thanks a lot.

```
header: 
  seq: 1554
  stamp: 
    secs: 370
    nsecs:  92000000
  frame_id: "odom"
child_frame_id: "base_footprint"
pose: 
  pose: 
    position: 
      x: -5.10870279874e-05
      y: 9.32454944017e-05
      z: 1.53271700887e-08
    orientation: 
      x: -9.63391262863e-08
      y: 2.13254059424e-09
      z: 7.15440031607e-07
      w: 1.0
  covariance: [1e-05, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 1e-05, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 1000000000000.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 1000000000000.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 1000000000000.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.001]
twist: 
  twist: 
    linear: 
      x: -3.3529998755e-06
      y: -4.87689774247e-09
      z: 0.0
    angular: 
      x: 0.0
      y: 0.0
      z: 3.47463370907e-09
  covariance: [0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0]
```
The completely error information is:
```
SUMMARY
========

PARAMETERS
 * /rosdistro: kinetic
 * /rosversion: 1.12.17
 * /use_sim_time: True

NODES
  /
    cartographer_node (cartographer_ros/cartographer_node)
    cartographer_occupancy_grid_node (cartographer_ros/cartographer_occupancy_grid_node)
    rviz (rviz/rviz)

ROS_MASTER_URI=http://localhost:11311

process[cartographer_node-1]: started with pid [6815]
process[cartographer_occupancy_grid_node-2]: started with pid [6816]
[ INFO] [1609163640.381360868]: I1228 21:54:00.000000  6815 configuration_file_resolver.cc:41] Found '/home/td/ros/catkin_ws/src/ros_basic_tutorials/mbot_slam/mbot_navigation/config/lidar.lua' for 'lidar.lua'.
[ INFO] [1609163640.381767062]: I1228 21:54:00.000000  6815 configuration_file_resolver.cc:41] Found '/home/td/slam/cartographer_ws/install_isolated/share/cartographer/configuration_files/map_builder.lua' for 'map_builder.lua'.
[ INFO] [1609163640.381800245]: I1228 21:54:00.000000  6815 configuration_file_resolver.cc:41] Found '/home/td/slam/cartographer_ws/install_isolated/share/cartographer/configuration_files/map_builder.lua' for 'map_builder.lua'.
[ INFO] [1609163640.381845676]: I1228 21:54:00.000000  6815 configuration_file_resolver.cc:41] Found '/home/td/slam/cartographer_ws/install_isolated/share/cartographer/configuration_files/pose_graph.lua' for 'pose_graph.lua'.
[ INFO] [1609163640.381870932]: I1228 21:54:00.000000  6815 configuration_file_resolver.cc:41] Found '/home/td/slam/cartographer_ws/install_isolated/share/cartographer/configuration_files/pose_graph.lua' for 'pose_graph.lua'.
[ INFO] [1609163640.381987745]: I1228 21:54:00.000000  6815 configuration_file_resolver.cc:41] Found '/home/td/slam/cartographer_ws/install_isolated/share/cartographer/configuration_files/trajectory_builder.lua' for 'trajectory_builder.lua'.
[ INFO] [1609163640.382019373]: I1228 21:54:00.000000  6815 configuration_file_resolver.cc:41] Found '/home/td/slam/cartographer_ws/install_isolated/share/cartographer/configuration_files/trajectory_builder.lua' for 'trajectory_builder.lua'.
[ INFO] [1609163640.382065073]: I1228 21:54:00.000000  6815 configuration_file_resolver.cc:41] Found '/home/td/slam/cartographer_ws/install_isolated/share/cartographer/configuration_files/trajectory_builder_2d.lua' for 'trajectory_builder_2d.lua'.
[ INFO] [1609163640.382089075]: I1228 21:54:00.000000  6815 configuration_file_resolver.cc:41] Found '/home/td/slam/cartographer_ws/install_isolated/share/cartographer/configuration_files/trajectory_builder_2d.lua' for 'trajectory_builder_2d.lua'.
[... 90 lines cut ...]
```

The launch file is:
```
<launch>  
  
  <param name="/use_sim_time" value="true" />  

  <node name="cartographer_node" pkg="cartographer_ros"  
      type="cartographer_node" args="
          -configuration_directory $(find mbot_navigation)/config
          -configuration_basename lidar.lua"
      output="screen"> 
    <remap from="scan" to="scan" />
  </node>  

  <!-- cartographer_occupancy_grid_node -->
  <node pkg="cartographer_ros" type="cartographer_occupancy_grid_node"
        name="cartographer_occupancy_grid_node" 
        args="-resolution 0.05" />

  <node name="rviz" pkg="rviz" type="rviz" required="true"  
        args="-d $(find cartographer_ros)/configuration_files/demo_2d.rviz" />  
</launch>
```

MichaelGrupp, 2021-01-11:
Your sensor data looks broken. `inf` values in odometry and negative laser readings don't make sense.

TouchDeeper, 2021-01-11:
@MichaelGrupp 
Thanks for your reply.
I also run the example of Deutsches Museum bag as said in the EDIT part. There is also an error:`F1229 10:32:34.000000 4231 msg_conversion.cc:135] Check failed: msg.range_min >= 0.f (-2.35183 vs. 0)`.
Could you please provide some advice? Thanks a lot.

MichaelGrupp, 2021-01-11:
As I said, negative laser readings don't make sense. I can't help you with debugging why your sensor driver publishes these values.

TouchDeeper, 2021-01-12:
@MichaelGrupp Thanks for your reply.
But the Deutsches Museum bag is the demo bag provide by cartographer ros [tutorial](https://google-cartographer-ros.readthedocs.io/en/latest/demos.html#id1). It should not have this problem:`F1229 10:32:34.000000 4231 msg_conversion.cc:135] Check failed: msg.range_min >= 0.f (-2.35183 vs. 0)`

pg95cm, 2022-07-14:
Hello, I have the same issue. Any updates on this?

laidvi, 2023-02-16:
I face the same issue. It happens when I play the ROS bags downloaded from the websites as well as when I use the /odom topic from Gazebo. Any updates or possible causes?