cartographer-project/cartographer_ros #1581: catkin_make_isolated --install --use-ninja --pkg cartographer_ros   error

charon-cheung, 2020-12-31:
I didn't encounter this error before, following is log 

```sh
 Installing: /home/xiaoqiang/Documents/ros/install_isolated/share/cartographer_ros/launch
-- Installing: /home/xiaoqiang/Documents/ros/install_isolated/share/cartographer_ros/launch/software_institute.launch
-- Up-to-date: /home/xiaoqiang/Documents/ros/install_isolated/share/cartographer_ros/launch/\
CMake Error at cmake_install.cmake:142 (file):
  file INSTALL cannot set permissions on
  "/home/xiaoqiang/Documents/ros/install_isolated/share/cartographer_ros/launch/\"

FAILED: cd /home/xiaoqiang/Documents/ros/build_isolated/cartographer_ros && /usr/bin/cmake -P cmake_install.cmake
ninja: build stopped: subcommand failed.
<== Failed to process package 'cartographer_ros':
  Command '['/home/xiaoqiang/Documents/ros/install_isolated/env.sh', 'ninja', 'install']' returned non-zero exit status 1

Reproduce this error by running:
==> cd /home/xiaoqiang/Documents/ros/build_isolated/cartographer_ros && /home/xiaoqiang/Documents/ros/install_isolated/env.sh ninja install

Command failed, exiting.
```
why is `install_isolated/share/cartographer_ros/launch/\`  created, very strange. How to solve this, I remove the launch folder, but it didn't work.

MichaelGrupp, 2021-01-11:
Did you check the permissions of your Linux user account for these directories?

charon-cheung, 2021-01-12:
> Did you check the permissions of your Linux user account for these directories?

I did. But never mind, I'm using the other computer, so I close this issue

closed 2021-01-12 by charon-cheung