cartographer-project/cartographer_ros #1582: ROS2 foxy: pure_localization=true, using map_server load a map, but cartographer still keep publish map

ipa-rwu, 2021-01-03:
Hello,

I am trying to use pure localization in ROS2 foxy. But catorgrapher still keep updating map even though I set pure_localization=true.
Basicly I am using the robot I built. It only use Lidar and doesn't have odom. 
As you can find [Config for Slam](https://github.com/ipa-rwu/minion_stack_ros2/blob/main/minion_cartographer/config/minion_laser_only.lua), I can create a map (not so good but can be used I think)

Based on [tuning guide for pure-localization-in-a-given-map](https://google-cartographer-ros.readthedocs.io/en/latest/tuning.html#pure-localization-in-a-given-map), and similar issues as #1403, #1258, #1031
I tuned the param as followed:
```
TRAJECTORY_BUILDER.pure_localization = true

POSE_GRAPH.optimize_every_n_nodes = 2
POSE_GRAPH.constraint_builder.sampling_ratio = 0.05
POSE_GRAPH.constraint_builder.min_score = 0.65
POSE_GRAPH.global_constraint_search_after_n_seconds = 20.
POSE_GRAPH.constraint_builder.ceres_scan_matcher.ceres_solver_options.max_num_iterations = 6
MAP_BUILDER.num_background_threads = 8
POSE_GRAPH.global_sampling_ratio = 0.003
```
I use nav_map_sever to load map, but I found the catrographer was still publishing to /map. 

Any help would be appreciated.
Thanks.

MichaelGrupp, 2021-01-11:
Probably you are running the `cartographer_occupancy_grid_node` in addition to the `cartographer_node`. The occupancy grid node publishes to `/map`. If you don't want anything on `/map` by this node, you need to remove it from your launch.

ipa-rwu, 2021-01-11:
@MichaelGrupp Thanks you very much for your advice. I will try out and come back to you (Probably this weekends)

nedo99, 2021-03-02:
Hi,

I was wondering how did you manage to run cartographer with foxy. Which launch files were you using? The ones that are in the `ros2-dashing-1.0.0` branch are still for ROS1, not ported to ROS2? When you use them with foxy, you get an invalid launch file error. I guess you used the ones here: https://github.com/ipa-rwu/minion_stack_ros2/tree/main/minion_cartographer/launch.

Are there default examples for cartographer for ROS2 somewhere that one could use for testing?

Regards,
Nedim