cartographer-project/cartographer_ros #1583: Robot locate fails in similar environmnet when mapping

charon-cheung, 2021-01-12:
I'm using latest version. My robot has a 2D lidar and odom. I'm actually using cartographer to expand map, like this [issue](https://github.com/cartographer-project/cartographer_ros/issues/1358), my launch file:
```
<launch>
  <node name="cartographer_node" pkg="cartographer_ros"
      type="cartographer_node" args="
          -configuration_directory $(find cartographer_ros)/configuration_files
          -configuration_basename localization.lua
          -load_state_filename /home/beerobot/carto_ws/maps/inner/4.pbstream"
      output="screen">
  <remap from="scan" to="/scan_rectified" />
  <remap from="odom" to="/odom" />
  </node>

  <node name="cartographer_occupancy_grid_node" pkg="cartographer_ros"
      type="cartographer_occupancy_grid_node" args="-resolution 0.05" />
</launch>
```

In general environment, mapping is fine, but in similar one, it definitely fails, like this [video](), but if I set `POSE_GRAPH.optimize_every_n_nodes = 0`, it wouldn't fail, I can get a completed map, but how to solve the problem once for all ? I think this issue is similar to [this one](https://github.com/cartographer-project/cartographer/issues/1441)

charon-cheung, 2021-02-03:
problem solved, closing this issue

closed 2021-02-03 by charon-cheung

SillyEndless, 2021-02-24:
Hi,Can you share the solution to us?