cartographer-project/cartographer_ros #1586: Switch Operation Mode during runtime

Tristan9497, 2021-01-14:
Hi,
is it possible to switch the operation mode of cartographer during runtime (from SLAM to localization only)?
I am using it to Map a road-like environment and want to switch to localization only after the robot has completed its first loop to minimize the computation cost.

Thanks in advance
Tristan

johuber, 2021-01-14:
You cannot change any parameters during run-time so you cannot change mode.

You'll probably want to export the pbstream and then restart cartographer in pure localization mode. In pure localization mode you probably want to use a different parameter set for optimal performance.

Tristan9497, 2021-01-14:
Thanks for the quick answer!
I'll take a look at that. It unfortunate that i cant change the mode, because i guess during restart of cartographer every tf transform will be dropped which in return will generate a lot of exceptions.

closed 2021-01-20 by johuber