cartographer-project/cartographer_ros #1587: Not able to generate map with google cartographer

VigneshS1993, 2021-01-14:
I collected sensor data such as imu, tf, odom, scan and tf_static data from gazebo simulation with a turtlebot. When I generate the map using gmapping with this bag file, I am able to get a map but when i run it with cartographer, I am not that successful. The generated does not have boundaries and has a lot of uncertainties. I have atached the pic in the github link as well.

->`rosbag_validate` gives some errors which I dont understand completed. The report is also in the github link here below:
  https://github.com/VigneshS1993/cartographerIssue/blob/main/cartographer_rosbag_validate_Result

- > The configuration, launch files and the default URDF file is also given in the same repository below: 
-  https://github.com/VigneshS1993/cartographerIssue

-> The link to download the bag file is here.
https://www.dropbox.com/s/b92dgwob9dlg6i9/simbag1_2021-01-14-16-37-31.bag?dl=0

The sample image generated by me is also uploaded here.

[image]

VigneshS1993, 2021-01-15:
And if i make the tracking frame as imu_link I get an error like the following:

[FATAL] [1610671844.641149382]: F0115 01:50:44.000000 23541 sensor_bridge.cc:136] Check failed: sensor_to_tracking->translation().norm() < 1e-5 The IMU frame must be colocated with the tracking frame. Transforming linear acceleration into the tracking frame will otherwise be imprecise.

eithwa, 2021-08-18:
[image]
I think there is something wrong with your imu so I stop using it
```
TRAJECTORY_BUILDER_2D.use_imu_data = true
TRAJECTORY_BUILDER_2D.use_online_correlative_scan_matching = true
```
and edit this from 10 to 1
```
TRAJECTORY_BUILDER_2D.num_accumulated_range_data = 1
```