cartographer-project/cartographer_ros #1589: mosaic of my 2d map boundary

panecho, 2021-01-20:
@gaschler 
thank you for this great work. I got some problem when try to use it.  Below is my step:
1 Here is a [ branch](https://github.com/panecho/cartographer_ros) of cartographer_ros I used
2 and cartographer_rosbag_validate [output](https://github.com/panecho/cartographer_ros/blob/master/questions_and_answers_files/artographer_rosbag_validate_output.txt)
3 and my launch [file](https://github.com/panecho/cartographer_ros/blob/master/cartographer_ros/launch/my_demo_backpack_2d.launch), [urdf](https://github.com/panecho/cartographer_ros/blob/master/cartographer_ros/urdf/my_backpack_2d.urdf)
4 and [rosbag](https://drive.google.com/file/d/17ptm3EuqMmC-4zhQ547XoUqqO8Z8DgdR/view?usp=sharing) I used.

My problem is that boundary of 2d OccupancyGrid map from my own data is mosaic, not as smooth as the map generated from **cartographer_paper_deutsches_museum.bag**, comparison show as below, right part is the problem map:   
[image]

what cause the mosaic and what I can do to make the map better?

MichaelGrupp, 2021-01-25:
Looks like an expected pixel discretization error to me. Did you try a finer resolution?

panecho, 2021-02-04:
> Looks like an expected pixel discretization error to me. Did you try a finer resolution?

Yes, Improving the resolution is helpful to reduce pixel discretization error But not the serrate boundary going down step by step to the one direction. like a wall represented by (a) showed below instead of (b) 
[image]

MichaelGrupp, 2021-03-15:
The discretization also depends on the angle. This is a natural limitation of any pixel image (you will have the same effect if you draw two lines in similar angles in a drawing program at low resolution).

closed 2021-03-15 by MichaelGrupp