cartographer-project/cartographer_ros #1590: Landmarks not processed

Niutels, 2021-01-26:
Hi,

I can't figure out how to use a landmark to localize my robot, could you help me figure what I am doing wrong?

What I can do:
1. Generate a pbstream map and feed it a QR-code landmark recognized in the environment.
2. Load the map while in pure localization mode.
3. See the marker /landmark_poses_list with rviz, at the right spot in the registered map.
4. Detect the landmark again while running in pure localization mode, I am able to see it at the correct pose in rviz too .

What I can't do: have cartographer localize my robot using this recognized landmark. 

Thank you!

johuber, 2021-01-26:
I don't quite understand which part fails above, what you describe sounds right. Are you failing to localize the robot in general or do you just feel like the landmarks are not being considered?

Are you trying to use landmarks only? Or are you trying to use landmarks for initial localization (kidnapped-robot)?

Niutels, 2021-01-26:
Thanks for your reply, I feel like the landmarks are not being considered as I want to use them for initial localization.

johuber, 2021-01-26:
I have also experimented with using them for an initial localization a while ago and came to the conclusion that it does not work.

Landmarks only seem to be considered once the robot is localized.

You'd have to dig pretty deep into the code to verify that, I think it would also be a welcome addition to the community if we could use it like that in case you have time.

@wohe , since you were involved in adding the implementation, do you remember why this is the case and if this is expected?

scenicpastures, 2021-04-08:
I have found the same behaviour i.e. landmarks not impacting time to localize. See https://github.com/cartographer-project/cartographer/issues/1813

JosefGst, 2023-03-24:
I succeeded initializing the robots location using landmarkers. What I did was simply increaseing the values for **translation_weight** and **rotation_weight** of my /landmark publisher. A value of 1e2 respectively worked for me. After startup it'll jump to the correct position if it dtects a landmark.