cartographer-project/cartographer_ros #1595: Current state of Continuing, Life-Long and Dynamic SLAM features

imstevenpmwyca, 2021-02-12:
Hello!
We have been using the 2D cartographer_ros package for our industrial mobile robot for the past few months. Right now we are interested in three main features: Continuing mapping (1st mapping session, save, load, continue 2nd mapping session, refine/remap zones), Life-Long Mapping and Localization (run slam permanently with bounded memory and cpu load) and robustness/adaptability to changing environments (specifically for low/semi-static/slow dynamics). 

After reviewing the available documentation and reading the relevant repository's issues, this is my understanding/questions of the current state of these three features:

- Continuing Mapping: This tags indicate that this feature is implemented (#1115 and #1507), but it is confusing how this works and documentation is missing. Would you please indicate how to properly use this feature?
- Life-Long Mapping and Localization: This tag and the milestone created indicate an initial interest in this feature (/cartographer-project/cartographer#197) back in 2017, but there isn't any documentation and it is unclear if it ever was ready for prime time. Is this currently implemented? If yes, could you please indicate how to use it? If not, what is the work done/left? Also, there are multiple discussions about a trimmer for removing data from overlapping sub-maps, I know this is related but I'm missing the bigger picture of the feature.
- Changing environments:  The Life-long feature (if implemented) could deal with this, but maybe this functionality is done (or could be) in a separate way. I couldn't find any information about this, so I would appreciate some clarification of the current capabilities.

I am asking all of this because I'm planning to work/improve these features for the following months, so I would like to know their current status.
Thank you very much for your time and consideration!

3473f, 2021-02-23:
> Continuing Mapping: This tags indicate that this feature is implemented (#1115 and #1507), but it is confusing how this works and documentation is missing. Would you please indicate how to properly use this feature?

Last time I used it (back in November/December last year) it worked like this: End the current trajectory, save .pbstream and end the current session. In the new session you should load the .pbstream and start a new trajectory with an initial pose. It should continue mapping on the new trajectory while the old one will be frozen.

imstevenpmwyca, 2021-03-02:
Thanks for the reply @3473f
Let me check if I get it right:
1. In the first mapping session, start the first trajectory calling the /start_trajectory service
2. Build the first map
3. End the first trajectory calling the /finish_trajectory service
4. Get the .pbstream file 
5. In the second mapping session, load the .pbstream file when launching the cartographer_node with the -load_state_filename argument
6. Start the second trajectory calling the /start_trajectory service with an initial pose
7. Build the second map
8. End the second trajectory calling the /finish_trajectory service
9. Get the .pbstream with the merged maps

I tried it like this, but when I try to start a trajectory the first time (step 1.) from terminal I get the following:
`ERROR: transport error completing service call: unable to receive data from sender, check sender's logs for details`
Do I have to setup something before trying it?

3473f, 2021-03-02:
I am unable to reproduce the error message you are getting, but you don't need to manually start a trajectory anyway. The Cartographer node should start the new trajectory by default. On launch you can see the following in your terminal
> [ INFO] [1614686728.209720608]: I0302 13:05:28.000000  3239 map_builder_bridge.cc:135] Added trajectory withI ID '0'.

starting a new trajectory using the same topics without terminating trajectory 0 first will result in an error.

> [ERROR] [1614687712.995527815]: E0302 13:21:52.000000  5157 node.cc:484] Topic name [scan] is already used.
> [ERROR] [1614687712.995608620]: E0302 13:21:52.000000  5157 node.cc:598] Topics are already used by another trajectory.

imstevenpmwyca, 2021-03-02:
You were right, I didn't need to start manually a trajectory in step 1.

Now I am stuck in step 5. When I launch the cartographer node I get:
`[FATAL] [/cartographer_node] [2021-03-02 13:07:16]: F0302 14:07:16.000000 1349925 proto_stream.cc:76] Check failed: in_.good() Failed to open proto stream '~/home/continuingtest.pbstream'.`

I an launching the node as follows, and I confirm the .pbstream exist in the directoy.
`<node name="cartographer_node" pkg="cartographer_ros"
          type="cartographer_node" args="
              -configuration_directory $(find package)/config
              -configuration_basename config.lua
              -load_state_filename ~/home/continuingtest.pbstream"
          output="screen" />`

Thanks for the help!

imstevenpmwyca, 2021-03-02:
Ok, I got the Continuing mapping feature working. 
These are the steps:

1. Start the mapping session as normal with cartographer_node
2. Build the first map
3. End the first trajectory calling the /finish_trajectory service
4. Get the .pbstream file calling the /write_state service
5. When starting the next mapping session, load the .pbstream file when launching the cartographer_node with the -load_state_filename argument
6. ~Start the second trajectory calling the /start_trajectory service with an initial pose~ The node will start a new trajectory automatically (`[ INFO] [/cartographer_node] [2021-03-02 13:51:09, 1970-01-01 00:08:09]: I0302 14:51:09.000000 1658686 map_builder_bridge.cc:135] Added trajectory with ID '1'.`)
7. Build the second map
8. End the second trajectory calling the /finish_trajectory service
9. Get the .pbstream with the merged maps and so on

The feature seems to be working both for exploration and exploitation of the environment. I have to further test with exploitation when the previous map has changed.

imstevenpmwyca, 2021-03-02:
For the Life-long mapping feature,
I am having this problem /cartographer-project/cartographer#1816
Have anyone use the submaps trimmer before?

GrantZheng86, 2021-04-05:
Hi @imstevenpmwyca, 
Have you ever encountered a trajectory accumulation issue when you update your map? I was basically doing exactly the same thing as you wrote above (with a bash script), and I was able to get the map to update. But whenever I call the "/finish_trajectory" service, the terminal did give me a 0 return code, but the trajectory does not disappear in RVIZ (even if I have removed the "trajectory" topic and add it back on). Also, the newly added trajectory (generated by the roslaunch cartographer node) got saved when I call the "/write_state" service. So, the next time I relaunch the cartographer node, another trajectory will be added to the map. 

The trajectory, however, did disappear from RVIZ after a very long time (like 5 minutes) after I call /finish_trajectory.

Does this happen to you, or this is just normal?

Thanks in advance for your help!

Grant

imstevenpmwyca, 2021-04-06:
Hello @GrantZheng86 ,

I am not sure if I understood completely your issue. But here are my comments:
1. I haven't encountered a trajectory accumulation issue. Is this a warning or error message?
2. When you call the finish trajectory service, the trajectory line from RVIZ doesn't go away, but the algorithm should stop processing data, you can quickly check this by looking the info messages displayed in terminal
3. Write state service will save all the trajectories, this is expected. If you don't want to add a new trajectory each time you roslaunch then you should use the argument '-start_trajectory_with_default_topics=false' in your launch file and then start a trajectory when you need it by calling the start trajectory service.

GrantZheng86, 2021-04-08:
Hi @imstevenpmwyca ,

Thank you so much for your response!

The trajectory accumulation isn't really a problem reported by cartographer. It's just that every time that I call /write_state, the just-finished trajectory will be added to the .pbstream file, and make the file a little bit larger. I guess it's just how /write_state works.

Thanks again for helping out!

imstevenpmwyca, 2021-04-14:
Dear @GrantZheng86,

What are you trying to accomplish? I am not sure if I'm understanding you.
For what I understand from your last comment, I would say:
1. Save the pbstream of the second mapping session (or second trajectory) with a different filename than the first one. So you have both the first mapping session pbstream and the first+second pbstream at hand.
2. If you want to have just the second pbstream, that is, if you want to decouple the second from the first one, I think you are better off just doing a new mapping session from scratch (without loading any previous pbstream).
3. If the problem is that the file is bigger each time you roslaunch (because a new trajectory is added) you can roslaunch without an active trajectory (as suggested in my previous comment), but like this you won't perform any mapping.
4. If the problem is that the file is bigger each time you write_state, then this is the expected behavior **if** a new trajectory was added. 
5. You don't really need to call write_state for the mapping to work. The write_state call is just for saving the pbstream file so you can use it later, like for continuing mapping. Even before calling finish_trajectory you should be able to see the map updating but after the finish_trajectory service is called (and the queue_elements are processed) the map won't be updated anymore with the sensor measurements.

GrantZheng86, 2021-04-23:
@imstevenpmwyca 
I've got what you mean, and it really helped me in updating the map! Thanks a lot

closed 2021-06-15 by imstevenpmwyca

GrantZheng86, 2021-06-29:
Hi @imstevenpmwyca,

Sorry to bother you again. When you are updating your map, are you keeping all the trajectories or you are overwrite the original map with a new one?

Thanks!