cartographer-project/cartographer_ros #1597: How to get the IMU calibration values from Cartographer?

juliajing72, 2021-03-02:
Hi, I am still new to Cartographer. and I am trying to find out how to use this to help the calibration between the LiDAR and IMU as suggested on the doc. 
I have enabled POSE_GRAPH.optimization_problem.log_solver_summary. However, I am still unsure where the IMU new outputs are?

swarmt, 2021-03-08:
Also interested in this

juliajing72, 2021-04-13:
> Also interested in this

Hi, I added the below and the log file was able to print out translation and rotation residuals as well as IMU corrections (in quaternion). But I'm a bit confused by why there is only one residual, as there are 3 translational parameters. 
POSE_GRAPH.constraint_builder.log_matches = true 
POSE_GRAPH.optimization_problem.log_solver_summary = true  -- calibrate IMU?
POSE_GRAPH.optimization_problem.use_online_imu_extrinsics_in_3d = true