cartographer-project/cartographer_ros #1600: Issues with OS1-128 lidar and Tinkerforge Brick v2.0 IMU

Jaetriel, 2021-03-11:
Hey, guys, I am pretty new to cartographer and ROS in general so i feel like the issues I'm having are probably something small I am missing. I've got an ouster lidar sensor and a Brick v2 IMU that I wanna connect to get real-time scanning of my surroundings with the lidar and use the IMU to help know it's orientation and location. I've taken the backpack_3d example config files and changed them a bit to match my setup but the issues I am having are that I'm not sure if the IMU data is registered at all, I dont see any axes moving or any part of the tf moving when I move the IMU, the virtual robot in RViz stays in one place. Also, I've managed to get a point cloud from the lidar showing but eventually I would need it to build a map and model of my surroundings whereas now I just see what the lidar is currently scanning when I move it around. I'll attach the configuration files that I'm using and the output of the validate command on a bag i recorded of the lidar and IMU. I can see from the output that the sensors need to be synced better, and I most probably have not set up my config files correctly for my needs. If you guys have any suggestions on what I'd need to fix or at least could point me to resources where I can learn more about ROS and cartographer so i can set it up on my own that would be greatly appreciated.

Output from rosbag_validate:

https://gist.github.com/Jaetriel/3f458f9088c3c91b4384636adcf50920

modified backpack_3d.lua file:

[image]

modified launch file: 

[image]

and my URDF file: 

[image]

johuber, 2021-03-12:
we cannot support individual tuning here, but have you looked at this guide: https://google-cartographer-ros.readthedocs.io/en/latest/tuning.html?highlight=pure%20localization#pure-localization-in-a-given-map

also: if the IMU is really not registered cartorapher would not output any data at all and there would be `ROS_WARN` output stating, that it is waiting to receive IMU data. That the robot pose does not move although you move the IMU could make sense, if the LIDAR stays stable and it still gets perfect matches there..

In terms of visualization: maybe you want to visualize the "submaps" of cartographer..

I hope this helps you to move on.

Jaetriel, 2021-03-16:
hey, thanks for the comment. I managed to get the sensor data and IMU data working together somewhat correctly, but now the submap that is visualized in RViz is really really blurry and low res and is not really ver correct I just see circles basically around my point cloud. I suppose that is mainly due to some wrong lua config settings that I am not setting correctly? [This recent rosbag_validate gist](https://gist.github.com/Jaetriel/69bafab147214a69ca3a85d3eb6eb8d8) tells me I have large gaps in the data from both IMU and lidar but I'm not sure what more I can do to remedy that I've used ptp time sync to get them both synchonized

soorajanilkumar, 2021-03-29:
Hi @Jaetriel ,
From the rosbag validate, I see that the IMU topic frequency is very low to for a 3D SLAM. 
Cartographer recomments 200 Hz minimum. (So try for a min 100 Hz at the very least. ). 

Also, for a quick trial, try
```
TRAJECTORY_BUILDER_3D.use_online_correlative_scan_matching = true
```
This is very processor intensive and should not be used for a 3D SLAM but is worth a try.