cartographer-project/cartographer_ros #1619: cartographer cannot run after setting "publish_tracked_pose=true" in eloquent

ninnyyan1993, 2021-05-04:
I am using ros2 eloquent version, as well as cartographer, I want to let the cartograher node publish 'pose' data to compare with ground truth data, so I add "publish_tracked_pose" in file "/$HOME/src/turtlebot3/turtlebot3_cartographer/config/turtlebot3_lds_2d.lua".
BTW, I start cartographer using "ros2 launch turtlebot3_cartographer cartographer.launch.py use_sim_time:=True", which use "turtlebot3_lds_2d.lua" configuration. 

Accoring to the official documentation here.["configure tracked_pose"](https://google-cartographer-ros.readthedocs.io/en/latest/ros_api.html#cartographer-node)
In "turtlebot3_lds_2d.lua", add "publish_tracked_pose_msg = true," in "options".

After adding, my lua file is as below:

`-- Copyright 2016 The Cartographer Authors
--
-- Licensed under the Apache License, Version 2.0 (the "License");
-- you may not use this file except in compliance with the License.
-- You may obtain a copy of the License at
--
--      http://www.apache.org/licenses/LICENSE-2.0
--
-- Unless required by applicable law or agreed to in writing, software
-- distributed under the License is distributed on an "AS IS" BASIS,
-- WITHOUT WARRANTIES OR CONDITIONS OF ANY KIND, either express or implied.
-- See the License for the specific language governing permissions and
-- limitations under the License.

-- /* Author: Darby Lim */

include "map_builder.lua"
include "trajectory_builder.lua"

options = {
  map_builder = MAP_BUILDER,
  trajectory_builder = TRAJECTORY_BUILDER,
  map_frame = "map",
  tracking_frame = "imu_link",
  published_frame = "odom",
  odom_frame = "odom",
  provide_odom_frame = false,
  publish_frame_projected_to_2d = true,
  use_odometry = true,
  use_nav_sat = false,
  use_landmarks = false,
  num_laser_scans = 1,
  num_multi_echo_laser_scans = 0,
  num_subdivisions_per_laser_scan = 1,
  num_point_clouds = 0,
  lookup_transform_timeout_sec = 0.2,
  submap_publish_period_sec = 0.3,
  pose_publish_period_sec = 5e-3,
  trajectory_publish_period_sec = 30e-3,
  rangefinder_sampling_ratio = 1.,
  odometry_sampling_ratio = 1.,
  fixed_frame_pose_sampling_ratio = 1.,
  imu_sampling_ratio = 1.,
  landmarks_sampling_ratio = 1.,
  publish_tracked_pose = true,
}

MAP_BUILDER.use_trajectory_builder_2d = true

TRAJECTORY_BUILDER_2D.min_range = 0.12
TRAJECTORY_BUILDER_2D.max_range = 3.5
TRAJECTORY_BUILDER_2D.missing_data_ray_length = 3.
TRAJECTORY_BUILDER_2D.use_imu_data = true
TRAJECTORY_BUILDER_2D.use_online_correlative_scan_matching = true 
TRAJECTORY_BUILDER_2D.motion_filter.max_angle_radians = math.rad(0.1)

POSE_GRAPH.constraint_builder.min_score = 0.65
POSE_GRAPH.constraint_builder.global_localization_min_score = 0.7

-- POSE_GRAPH.optimize_every_n_nodes = 0

return options`

then RUN "ros2 launch turtlebot3_cartographer cartographer.launch.py use_sim_time:=True", the cartographer node is not started, cannot find the node in rqt(before change the setting in options, the cartographer node can be started successfully), and I got this error in terminal:

`[cartographer_node-1] F0503 09:50:53.577739  6078 lua_parameter_dictionary.cc:410] Check failed: 1 == reference_counts_.count(key) (1 vs. 0) Key 'publish_tracked_pose' was used the wrong number of times.
[cartographer_node-1] [FATAL] [cartographer_ros]: F0503 09:50:53.000000  6078 lua_parameter_dictionary.cc:410] Check failed: 1 == reference_counts_.count(key) (1 vs. 0) Key 'publish_tracked_pose' was used the wrong number of times.
`

Can anyone help me on this question?

BTW, in another page of official documentation [here](https://google-cartographer-ros.readthedocs.io/en/latest/configuration.html), to publish "tracked_pose" topic, we are required to add "publish_tracked_pose_msg=true" instead of change "publish_tracked_pose=true" in [previous documentation](https://google-cartographer-ros.readthedocs.io/en/latest/ros_api.html#cartographer-node), which one should we use?

johuber, 2021-05-04:
Thank you for reporting the bug in the documentation. The correct one to use is `publish_tracked_pose`, I will fix this on the referenced page.

The error above indicates though, that either you did not add the parameter `publish_tracked_pose = true` or that you are not loading this file. Did you verify that you are loading the correct parameter file and that you are using the correct parameter name (without `_msg`) ?

ninnyyan1993, 2021-05-04:
> Thank you for reporting the bug in the documentation. The correct one to use is `publish_tracked_pose`, I will fix this on the referenced page.
>
> The error above indicates though, that either you did [...]

Thank you for replying. About the 2 questions you asked. 
1. If I did not added "publish_tracked_pose" in the file, there will not be an error which indicates "Key 'publish_tracked_pose' was used the wrong number of times. [cartographer_node-1] [FATAL]", am I right?
2. If I edit the wrong file, as I have described in the question, I think there will be no error that I cannot find "cartographer node" in rqt graph, because before I add this line, the cartographer node runs well, haha.

johuber, 2021-05-05:
This is a bit odd, because the parameter is optional and I don't fully understand how it can come to this exception yet.

can you try if you have similar behavior if you set another optional parameter (e.g. `publish_to_tf = true`) ?

tim-fan, 2021-05-06:
Do you know what version of cartographer is released in eloquent?

I'm hitting the same error in melodic, and I think it is because the released version is v 1.0, which was tagged 3 years ago,  before that parameter was introduced. 

I guess in my case a source-build is required, perhaps it is the same for you.

Looking at cartographer_ros branches, I see there is a `ros2-dashing` branch which looks quite close to master, and a `ros2-dashing-1.0.0` branch, which is only one commit ahead of `release-1.0`. If you're using the later branch that might explain why the parameter is unavilable.

I'm getting my cartographer version by:
```
apt-cache show ros-melodic-cartographer-ros | grep Version
Version: 1.0.0-1bionic.20210414.231737
```

ninnyyan1993, 2021-05-10:
@tim-fan Thank you so much for your reply and sorry for replying late,  I was working on something else. 
Let me have a  try! Will give feedback later. Thanks again.

ninnyyan1993, 2021-05-10:
> @tim-fan Thank you so much for your reply and sorry for replying late, I was working on something else.
> Let me have a try! Will give feedback later. Thanks again.

closed 2021-05-10 by ninnyyan1993

safdark, 2021-05-20:
I'm getting this same error too.

Version: 1.0.9003-1focal.20210423.021006

Is there a link somewhere that publishes mappings of version# -> commit# that I could sync my code to, so I could investigate the right codebase?