cartographer-project/cartographer_ros #1627: Receiving incorrect values from /tracked_pose

Vineet-Pandey, 2021-05-18:
Hi,

I am using Cartographer_3D  and I am able to generate the map of the an area without any issue, as shown below:
[image]
However, when I try getting the position using the /tracked_pose topic, I receive grossly incorrect values. Could someone please advise on what could be the issue?

johuber, 2021-05-19:
do you have a recording of these "grossly incorrect values"?
E.g. you could visualize the output of the tracked pose topic in RVIZ and share the result

Vineet-Pandey, 2021-05-19:
I have recorded the values published on the topic in a rosbag in the link below: 
[https://drive.google.com/file/d/13vC1m8AXMXXOxIYtPeMTOpcjJPJLL2Da/view?usp=sharing](url)

Below is the corresponding video:
https://user-images.githubusercontent.com/29963970/118837279-a6e9ac80-b892-11eb-89da-9dd5eff1aff3.mp4

johuber, 2021-05-19:
that's odd. could it be that there is something wrong in your TF tree? what frames did you configure in cartographer?

Vineet-Pandey, 2021-05-19:
So, I have configured an IMU frame (sensor_wgs84) and a lidar frame(livox_frame_3GGDHAD00100521) in an URDF , as below:
[https://drive.google.com/file/d/1LeH_awhOSlDsEVklbWySQvXHig-LeiyH/view?usp=sharing](url)
And my TF tree looks like:
[image]

Also, I don't know if this is of any significance in identifying the issue, but in the /map topic, the positions (info.origin.position) look much more reasonable.

johuber, 2021-05-20:
what did you set in the settings as your `tracking_frame`?

 The pose published is the pose of the tracking frame in the `map` frame.  [This line ](https://github.com/cartographer-project/cartographer_ros/blob/c06879b63567a78ef92b7d1fa79453839e36ffec/cartographer_ros/cartographer_ros/node.cc#L285) defines the pose that is then published.

Any ideas what could be wrong when you look at the code there?

Vineet-Pandey, 2021-05-20:
I have tried both ```base_link``` and ```sensor_wgs84``` (the imu_link) as the ```tracking_frame``` but both give the same o/p.
I don't see any issue around that particular line of code too. Let me dig deeper into this. Could you please leave this issue open for a while so that if I come up with a solution or a question I can post it here?

Vineet-Pandey, 2021-05-20:
Here is my finding, and I am unable to understand why Cartographer behaves this way:

When I run cartographer while getting the data from actual sensors in real-time (not from a rosbag), I get correct positions ( with a few drift errors which I think can be corrected through tuning). But, when I record the sensor readings in a rosbag and run it, the positions provided by Cartographer (/tracked_pose) are incorrect.

Is there some nuance in cartographer that I am missing?

johuber, 2021-05-21:
> Here is my finding, and I am unable to understand why Cartographer behaves this way:
>
> When I run cartographer while getting the data from actual sensors in real-time (not from a rosbag), I get [...]

I can think of a few things to explore:
- tf issue --> are you recording any `/tf` and `/tf_static` to your bag? is it possible that there are competing TFs coming from both the bag and cartographer?
- clock issue --> where is your clock coming from when you use the bag? running it with `--clock` or not?

Those are just the differences I can come up with between running from bags and running with live data.

Sorry, I don't have capacity right now to test it much myself.

closed 2021-05-27 by Vineet-Pandey