cartographer-project/cartographer_ros #1632: About TSDF lua file

kay01-kwon, 2021-05-24:
Hi, I'm working with a mobile manipulator which has four passive suspension and the robot execute SLAM.
To achieve the accurate positioning of the mobile robot e.g. within 5 mm, it is needed to secure the performance of localization and some paper mentions Truncated Signed Distance Field to make the accuracy within 5 mm. My question is... Are there any reference when it comes to TSDF lua file and I want to get the meaning of parameters precisely. Also, how do I have to set ceres scan matching in pose graph and trajectory 2d to improve the accuracy?

kdaun, 2021-05-24:
Hi,
unfortunately, the TSDF pipeline is still undocumented. I will aim to add documentation soon.
In my experience, TSDF can provide small accuracy benefits in real-world datasets in comparison to  occupancy grid. See a brief discussion in https://github.com/cartographer-project/cartographer_ros/issues/1058 and a more thorough one in my paper[(1)](https://www.sim.informatik.tu-darmstadt.de/publ/download/2019_daun_ssrr.pdf)
If you reach your target accuracy of 5mm will strongly depend on your hardware, sensors, calibration, compute and the operation environment.

[(1) Daun, Kevin, et al. "Large scale 2d laser slam using truncated signed distance functions." 2019 IEEE International Symposium on Safety, Security, and Rescue Robotics (SSRR). IEEE, 2019.](https://www.sim.informatik.tu-darmstadt.de/publ/download/2019_daun_ssrr.pdf)

kay01-kwon, 2021-05-25:
Yeah the paper I read is your one. My robot uses VLP16 lidar and odometry via encoders. And the mini cpu which launches SLAM and tf is Xavier. The documentation of cartographer mentions the requirements of the CPU which is i7 3rd generation. Can the Xavier CPU limitation be the problem?