cartographer-project/cartographer_ros #1636: Error while installing cartographer_ros

chinmaynehate, 2021-06-01:
Hey,

OS: Ubuntu 16.04
I ran all the commands as given in: https://google-cartographer-ros.readthedocs.io/en/latest/compilation.html for ubuntu 16.04.
When I ran the last command: catkin_make_isolated --install --use-ninja, I got the following error: 

```

avitra@nuc:~/cartographer_ws$ catkin_make_isolated --install --use-ninja
Base path: /home/avitra/cartographer_ws
Source space: /home/avitra/cartographer_ws/src
Build space: /home/avitra/cartographer_ws/build_isolated
Devel space: /home/avitra/cartographer_ws/devel_isolated
Install space: /home/avitra/cartographer_ws/install_isolated
~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
~~  traversing 4 packages in topological order:
~~  - cartographer (plain cmake)
~~  - cartographer_ros_msgs
~~  - cartographer_ros
~~  - cartographer_rviz
~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~

==> Processing plain cmake package: 'cartographer'
build.ninja exists, skipping explicit cmake invocation...
==> ninja build.ninja in '/home/avitra/cartographer_ws/build_isolated/cartographer/install'
ninja: no work to do.
==> ninja -j8 -l8 in '/home/avitra/cartographer_ws/build_isolated/cartographer/install'
[8/344] Building CXX object CMakeFiles/cartographer.sensor.internal.collator_test.dir/cartographer/sensor/internal/collator_test.cc.o
FAILED: /usr/bin/c++ -D__CLANG_SUPPORT_DYN_ANNOTATION__ -I/home/avitra/cartographer_ws/src/cartographer/cartographer -I. -I/home/avitra/cartographer_ws/src/cartographer -isystem /usr/src/gmock/include -isyste
...
...
...
/usr/local/include/absl/strings/string_view.h:487:3: error: ‘size_type’ does not name a type
   size_type find_last_not_of(char c, size_type pos = npos) const
   ^
/usr/local/include/absl/strings/string_view.h:491:10: error: ‘constexpr’ does not name a type
   static constexpr size_type kMaxSize =
          ^
/usr/local/include/absl/strings/string_view.h:491:10: note: C++11 ‘constexpr’ only available with -std=c++11 or -std=gnu++11
/usr/local/include/absl/strings/string_view.h:494:10: error: ‘constexpr’ does not name a type
   static constexpr size_type CheckLengthInternal(size_type len) {
          ^
/usr/local/include/absl/strings/string_view.h:494:10: note: C++11 ‘constexpr’ only available with -std=c++11 or -std=gnu++11
/usr/local/include/absl/strings/string_view.h:499:3: error: ‘size_type’ does not name a type
   size_type length_;
   ^
/usr/local/include/absl/strings/string_view.h:354:51: error: ‘npos’ was not declared in this scope
   string_view substr(size_type pos, size_type n = npos) const {
                                                   ^
/usr/local/include/absl/strings/string_view.h: In member function ‘void absl::string_view::remove_prefix(int)’:
/usr/local/include/absl/strings/string_view.h:313:5: error: ‘length_’ was not declared in this scope
     length_ -= n;
     ^
/usr/local/include/absl/strings/string_view.h: In member function ‘void absl::string_view::remove_suffix(int)’:
/usr/local/include/absl/strings/string_view.h:322:5: error: ‘length_’ was not declared in this scope
     length_ -= n;
     ^
In file included from /usr/local/include/absl/base/port.h:24:0,
                 from /usr/local/include/absl/base/macros.h:34,
                 from /usr/local/include/absl/strings/substitute.h:73,
                 from /home/avitra/cartographer_ws/src/cartographer/cartographer/transform/rigid_transform.h:26,
                 from /home/avitra/cartographer_ws/src/cartographer/cartographer/transform/transform.h:26,
                 from /home/avitra/cartographer_ws/src/cartographer/cartographer/sensor/rangefinder_point.h:24,
                 from /home/avitra/cartographer_ws/src/cartographer/cartographer/sensor/point_cloud.h:24,
                 from /home/avitra/cartographer_ws/src/cartographer/cartographer/sensor/point_cloud_test.cc:17:
/usr/local/include/absl/strings/string_view.h: In member function ‘absl::string_view absl::string_view::substr(int, int) const’:
/usr/local/include/absl/strings/string_view.h:355:9: error: ‘length_’ was not declared in this scope
     if (ABSL_PREDICT_FALSE(pos > length_))
         ^
In file included from /usr/local/include/absl/strings/ascii.h:59:0,
[... 43 lines cut ...]
     return substr(pos1, count1).compare(string_view(s, count2));
                                                              ^
/usr/local/include/absl/strings/string_view.h:144:7: note: candidate: absl::string_view::string_view()
 class string_view {
       ^
/usr/local/include/absl/strings/string_view.h:144:7: note:   candidate expects 0 arguments, 2 provided
/usr/local/include/absl/strings/string_view.h:144:7: note: candidate: absl::string_view::string_view(const absl::string_view&)
/usr/local/include/absl/strings/string_view.h:144:7: note:   candidate expects 1 argument, 2 provided
/usr/local/include/absl/strings/string_view.h: At global scope:
/usr/local/include/absl/strings/string_view.h:505:54: error: expected initializer before ‘noexcept’
 inline bool operator==(string_view x, string_view y) noexcept {
                                                      ^
/home/avitra/cartographer_ws/src/cartographer/cartographer/sensor/point_cloud_test.cc:91:1: error: expected ‘}’ at end of input
 }  // namespace cartographer
 ^
ninja: build stopped: subcommand failed.
<== Failed to process package 'cartographer': 
  Command '['ninja', '-j8', '-l8']' returned non-zero exit status 1

Reproduce this error by running:
==> cd /home/avitra/cartographer_ws/build_isolated/cartographer && ninja -j8 -l8

Command failed, exiting.
avitra@nuc:~/cartographer_ws$ 
```

aseemk98, 2025-03-20:
@chinmaynehate Were you able to find a fix for this?

chinmaynehate, 2025-03-20:
@aseemk98 

No, I did not work on it after this issue.