cartographer-project/cartographer_ros #1643: How to edit map data?

makeitjune, 2021-06-09:
Hi,
I'm beginner using cartographer.
I made a map file(.pgm) from pbstream file.
[image]

After that,I ran pure localization and it works.
So, i want to modify a map using external tool(gimp or some image edit tool) to add or remove data(obstacles, walls etc).

But if i edit .pgm file, how can i apply change thing to .pbstream?
As far as i know, cartographer save all SLAM data(trajectory, submap, pose_graph etc) to  .pbstream file.
Running pure_localization, I should load a .pbstream file. 

If i want to edit map data what i have to do?
Thank you.

johuber, 2021-06-09:
The pgm file is just there for visualization purposes and it is not possible to change the pbstream by editing the pgm file.
Developing a tool to edit map data would be a significant effort and there is no trivial way of doing it.

One "simpler" approach if you need to filter some obstacles and walls is already filtering the corresponding points in the original point cloud that you use to build the map.

makeitjune, 2021-06-15:
> The pgm file is just there for visualization purposes and it is not possible to change the pbstream by editing the pgm file.
> Developing a tool to edit map data would be a significant effort and [...]

> The pgm file is just there for visualization purposes and it is not possible to change the pbstream by editing the pgm file.
> Developing a tool to edit map data would be a significant effort and [...]

Thank you

closed 2021-06-15 by makeitjune

MigVega, 2022-11-27:
As you mentioned [makeitjune](https://github.com/makeitjune), you will need the original `.pbtream` map to use cartographer in pure localization mode.
I understand the need to edit the map and that doing it on `.pgm` is easier.

Besides what [johuber](https://github.com/johuber) suggested, you can also convert your  `.pgm`  again to `.pbtream` with the [Ogm2Pgbm package](https://github.com/MigVega/Ogm2Pgbm/tree/main/ogm2pgbm).

Additionally, if you wanna run the tool in a separate docker container, you can follow the instructions in the [readme](https://github.com/MigVega/Ogm2Pgbm).

Have fun, and enjoy the power of graph-based localization! :D