cartographer-project/cartographer_ros #1656: Error during installation

ManChrys, 2021-07-09:
Hello guys,

i follow the instructions he https://google-cartographer-ros.readthedocs.io/en/latest/compilation.html

but when im catkin_make_isolated --install --use-ninja i have the following error

```
collect2: error: ld returned 1 exit status
ninja: build stopped: subcommand failed.
<== Failed to process package 'cartographer_ros': 
  Command '['/home/geo/catkin_ws/install_isolated/env.sh', 'ninja', '-j16', '-l16']' returned non-zero exit status 1

Reproduce this error by running:
==> cd /home/geo/catkin_ws/build_isolated/cartographer_ros && /home/geo/catkin_ws/install_isolated/env.sh ninja -j16 -l16

Command failed, exiting.

Thanks in advance :D

georgflick, 2021-07-15:
@ManChrys can you provide more infos? Interesting would be the lines before `collect2: error: ld returned 1 exit status`
Thanks!

vargabalint92, 2021-11-10:
Hi @georgflick 

i have a similar error. here it is its last 1/3.

/usr/local/include/absl/meta/type_traits.h:450:1: error: expected unqualified-id before ‘using’
 using common_type_t = typename std::common_type<T...>::type;
 ^
/usr/local/include/absl/meta/type_traits.h:453:1: error: expected unqualified-id before ‘using’
 using underlying_type_t = typename std::underlying_type<T>::type;
 ^
/usr/local/include/absl/meta/type_traits.h:456:1: error: expected unqualified-id before ‘using’
 using result_of_t = typename std::result_of<T>::type;
 ^
/usr/local/include/absl/meta/type_traits.h:475:37: error: expected class-name before ‘{’ token
 struct IsHashable : std::false_type {};
                                     ^
/usr/local/include/absl/meta/type_traits.h:480:5: error: ‘enable_if_t’ is not a member of ‘absl’
     absl::enable_if_t<std::is_convertible<
     ^
/usr/local/include/absl/meta/type_traits.h:480:5: error: ‘enable_if_t’ is not a member of ‘absl’
/usr/local/include/absl/meta/type_traits.h:480:23: error: ‘is_convertible’ is not a member of ‘std’
     absl::enable_if_t<std::is_convertible<
                       ^
/usr/local/include/absl/meta/type_traits.h:480:23: note: suggested alternative:
In file included from /usr/include/eigen3/Eigen/Core:295:0,
                 from /localssd/balint/catkin_ws3/src/cartographer/cartographer/mapping/3d/hybrid_grid.h:26,
                 from /localssd/balint/catkin_ws3/src/cartographer/cartographer/mapping/internal/3d/scan_matching/precomputation_grid_3d.h:20,
                 from /localssd/balint/catkin_ws3/src/cartographer/cartographer/mapping/internal/3d/scan_matching/precomputation_grid_3d.cc:17:
/usr/include/eigen3/Eigen/src/Core/util/Meta.h:122:8: note:   ‘Eigen::internal::is_convertible’
 struct is_convertible
        ^
In file included from /usr/local/include/absl/memory/memory.h:34:0,
                 from /localssd/balint/catkin_ws3/src/cartographer/cartographer/mapping/3d/hybrid_grid.h:27,
                 from /localssd/balint/catkin_ws3/src/cartographer/cartographer/mapping/internal/3d/scan_matching/precomputation_grid_3d.h:20,
                 from /localssd/balint/catkin_ws3/src/cartographer/cartographer/mapping/internal/3d/scan_matching/precomputation_grid_3d.cc:17:
/usr/local/include/absl/meta/type_traits.h:481:18: error: ‘declval’ is not a member of ‘std’
         decltype(std::declval<std::hash<Key>&>()(std::declval<Key const&>())),
                  ^
/usr/local/include/absl/meta/type_traits.h:481:31: error: ‘hash’ is not a member of ‘std’
         decltype(std::declval<std::hash<Key>&>()(std::declval<Key const&>())),
                               ^
/usr/local/include/absl/meta/type_traits.h:481:45: error: ‘&’ cannot appear in a constant-expression
         decltype(std::declval<std::hash<Key>&>()(std::declval<Key const&>())),
                                             ^
/usr/local/include/absl/meta/type_traits.h:481:50: error: ‘declval’ is not a member of ‘std’
         decltype(std::declval<std::hash<Key>&>()(std::declval<Key const&>())),
                                                  ^
/usr/local/include/absl/meta/type_traits.h:481:76: error: a function call cannot appear in a constant-expression
         decltype(std::declval<std::hash<Key>&>()(std::declval<Key const&>())),
                                                                            ^
/usr/local/include/absl/meta/type_traits.h:481:77: error: a function call cannot appear in a constant-expression
         decltype(std::declval<std::hash<Key>&>()(std::declval<Key const&>())),
                                                                             ^
/usr/local/include/absl/meta/type_traits.h:482:20: error: wrong number of template arguments (3, should be at least 1)
         std::size_t>::value>> : std::true_type {};
                    ^
/usr/local/include/absl/meta/type_traits.h:475:8: note: provided for ‘template<class Key, class> struct absl::type_traits_internal::IsHashable’
 struct IsHashable : std::false_type {};
        ^
/usr/local/include/absl/meta/type_traits.h:482:28: error: expected unqualified-id before ‘>>’ token
         std::size_t>::value>> : std::true_type {};
                            ^
/usr/local/include/absl/meta/type_traits.h:492:10: error: expected type-specifier before ‘decltype’
       -> decltype(std::declval<std::hash<Key>>()(std::declval<Key const&>()));
          ^
/usr/local/include/absl/meta/type_traits.h:492:10: error: expected initializer before ‘decltype’
/usr/local/include/absl/meta/type_traits.h:497:15: error: ‘nullptr_t’ in namespace ‘std’ does not name a type
   static std::nullptr_t DoIt() {
               ^
/usr/local/include/absl/meta/type_traits.h: In function ‘void absl::type_traits_internal::AssertHashEnabled()’:
[... 2186 lines cut ...]
                   ^
/usr/local/include/absl/utility/utility.h:266:8: error: expected type-specifier before ‘decltype’
     -> decltype(utility_internal::apply_helper(
        ^
/usr/local/include/absl/utility/utility.h:266:8: error: expected initializer before ‘decltype’
/usr/local/include/absl/utility/utility.h:269:67: error: expected unqualified-id before ‘)’ token
             typename std::remove_reference<Tuple>::type>::value>{})) {
                                                                   ^
In file included from ./cartographer/mapping/proto/trajectory_node_data.pb.h:32:0,
[... cut ...]
                 from /localssd/balint/catkin_ws3/src/cartographer/cartographer/metrics/register.cc:19:
./cartographer/sensor/proto/sensor.pb.h:1286:30: error: expected ‘}’ before end of line
./cartographer/sensor/proto/sensor.pb.h:1286:30: error: expected declaration before end of line
ninja: build stopped: subcommand failed.
<== Failed to process package 'cartographer':
  Command '['ninja', '-j4', '-l4']' returned non-zero exit status 1

Reproduce this error by running:
==> cd /localssd/balint/catkin_ws3/build_isolated/cartographer && ninja -j4 -l4

Command failed, exiting.

georgflick, 2021-11-19:
Hi @vargabalint92 ,
which compiler, ubuntu and ros are you using? Supported is Ubuntu 20.04 (Focal) and ROS noetic

My output of CMake.
```
-- The C compiler identification is GNU 9.3.0
-- The CXX compiler identification is GNU 9.3.0
-- Check for working C compiler: gcc
-- Check for working C compiler: gcc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: g++
-- Check for working CXX compiler: g++ -- works
```

If CMake finds other compilers, you can change them by using the CC and CXX environment variable:
e.g.:
`export CC="gcc" CXX="g++"`

Hope that helps you to identify and solve the issue.