cartographer-project/cartographer_ros #1668: Tracked_pose Topic is not being published

impaidk, 2021-08-05:
I'm running the 2D Cartographer on my own recorded bag file. My cartographer creates submaps for the rosbag data. Now I want it to publish the tracked_pose topic so that i can evaluate my localization using some other script and comparing it to the GPS data.

In the cartographer_2d.lua config file, I have set the parameter 

> publish_tracked_pose = true

. As per the documentation, tracked_pose should be published if the above parameter is set to true. My cartographer creates the submaps without any errors. Can someone please direct me to right direction, is there some other parameter which the tracked_pose depends on? Thanks in advance.

If you have any questions of my setup, I would be happy to answer.

channing-prince, 2022-04-22:
I'm a user of Cartographer, As far as I know, this is the only parameter that enables it to publish the tracked_pose topic ,when I set this parameter to true , run the bag, then I use 
`rosrun rqt_topic rqt_topic`
 to check all the topic, the `/tracked_pose` is just there 

Have you re-compile the workspace or did you find the lua file through `roscd cartographer_ros/configuration_files/` and then edit it?
If you don't do that, the config file won't come into effect
[image]

PrathameshCOLLABINT, 2024-07-12:
> Hi Cartographer Team,
> 
> I'm running the 2D Cartographer on my own recorded bag file. My cartographer creates submaps for the rosbag data. Now I want it to publish the tracked_pose topic so that i can evaluate my localization using some other script and comparing it to the GPS data.
> 
> In the cartographer_2d.lua config file, I have set the parameter
> 
> > publish_tracked_pose = true
> 
> . As per the documentation, tracked_pose should be published if the above parameter is set to true. My cartographer creates the submaps without any errors. Can someone please direct me to right direction, is there some other parameter which the tracked_pose depends on? Thanks in advance.
> 
> If you have any questions of my setup, I would be happy to answer.
> 
> Regards, DP

hi can you mark here which documents are you refring i am new here finding learning source

this whould help me