cartographer-project/cartographer_ros #1669: Fixed Frame [map] does not exist

SuperrHaze, 2021-08-05:
**General discription:** Cartographer succeed to publish topic /map and submaps are shown in rviz, however, the Fixed Frame [map] does not exist, and the tf transform map->odom doesn't exit either.**I use robot_localization to publish /odom.**
**RVIZ:** 
[image]
**lua file:**
```
include "map_builder.lua"
include "trajectory_builder.lua"
 
options = {
    map_builder = MAP_BUILDER,
    trajectory_builder = TRAJECTORY_BUILDER,
    map_frame = "map",
    tracking_frame = "base_link",
    published_frame = "odom",
    odom_frame = "odom",
    provide_odom_frame = false,
    publish_frame_projected_to_2d = false,
    use_pose_extrapolator = true,
    use_odometry = true,
    use_nav_sat = false,
    use_landmarks = false,
    num_laser_scans = 1,
    num_multi_echo_laser_scans = 0,
    num_subdivisions_per_laser_scan = 1,
    num_point_clouds = 0,
    lookup_transform_timeout_sec = 0.2,
    submap_publish_period_sec = 0.3,
    pose_publish_period_sec = 5e-3,
    trajectory_publish_period_sec = 30e-3,
    rangefinder_sampling_ratio = 1.,
    odometry_sampling_ratio = 1.,
    fixed_frame_pose_sampling_ratio = 1.,
    imu_sampling_ratio = 1.,
    landmarks_sampling_ratio = 1.,
}
 
MAP_BUILDER.use_trajectory_builder_2d = true
 
TRAJECTORY_BUILDER_2D.submaps.num_range_data = 10
TRAJECTORY_BUILDER_2D.min_range = 0.3
TRAJECTORY_BUILDER_2D.max_range = 6.
TRAJECTORY_BUILDER_2D.missing_data_ray_length = 1.
TRAJECTORY_BUILDER_2D.use_imu_data = false
TRAJECTORY_BUILDER_2D.use_online_correlative_scan_matching = true
TRAJECTORY_BUILDER_2D.real_time_correlative_scan_matcher.linear_search_window = 0.1
TRAJECTORY_BUILDER_2D.real_time_correlative_scan_matcher.translation_delta_cost_weight = 10.
TRAJECTORY_BUILDER_2D.real_time_correlative_scan_matcher.rotation_delta_cost_weight = 1e-1

return options
```
**tf_tree:**
[image]
**rqt_graph:**
[image]
**rostopic list:**
[image]
**cartographer terminal:**
[image]
**launch file:**
```
<launch>
 
  <param name="/use_sim_time" value="true" />  
 
  <node name="cartographer_node" pkg="cartographer_ros"  
        type="cartographer_node" args="  
            -configuration_directory $(find cartographer_ros)/configuration_files  
            -configuration_basename delta_lidar.lua"  
        output="screen">  
    <remap from="scan" to="scan" />  
  </node>  
  
  <node name="cartographer_occupancy_grid_node" pkg="cartographer_ros"
      type="cartographer_occupancy_grid_node" args="-resolution 0.05" />
	  
  <node name="rviz" pkg="rviz" type="rviz" required="true"  
        args="-d $(find cartographer_ros)/configuration_files/demo_2d.rviz" />  
</launch>
```
Thank you for any suggestions！

moooeeeep, 2021-09-14:
I have the same isse currently. 

My current hypothesis is that the version of cartographer_ros that I am using is so old (the one distributed via the ros package repository), that it doesn't provide that functionality. For reference:

    Package: ros-melodic-cartographer-ros
    Version: 1.0.0-1bionic.20210505.045052
    Priority: extra
    Section: misc
    Maintainer: The Cartographer Authors <cartographer-owners@googlegroups.com>
    Installed-Size: 32,0 MB
    Depends: libboost-iostreams1.65.1, libboost-system1.65.1, libc6 (>= 2.27), libcairo2 (>= 1.6.0), libceres1, libgcc1 (>= 1:3.0), libgflags2.2, libgoogle-glog0v5, liblua5.2-0, libpcl-common1.8, libprotobuf10, libstdc++6 (>= 6), liburdfdom-world, libgflags-dev, libgoogle-glog-dev, libpcl-dev, ros-melodic-cartographer (>= 1.0.0), ros-melodic-cartographer-ros-msgs (>= 1.0.0), ros-melodic-eigen-conversions, ros-melodic-geometry-msgs, ros-melodic-message-runtime, ros-melodic-nav-msgs, ros-melodic-pcl-conversions, ros-melodic-robot-state-publisher, ros-melodic-rosbag, ros-melodic-roscpp, ros-melodic-roslaunch, ros-melodic-roslib, ros-melodic-sensor-msgs, ros-melodic-std-msgs, ros-melodic-tf2, ros-melodic-tf2-eigen, ros-melodic-tf2-ros, ros-melodic-urdf, ros-melodic-visualization-msgs
    Homepage: https://github.com/googlecartographer/cartographer_ros
    Download-Size: 4.014 kB
    APT-Manual-Installed: yes
    APT-Sources: http://packages.ros.org/ros/ubuntu bionic/main amd64 Packages
    Description: Cartographer is a system that provides real-time simultaneous localization and mapping (SLAM) in 2D and 3D across multiple platforms and sensor configurations.
     This package provides Cartographer's ROS integration.

When I try to set the parameters [`publish_to_tf` and `publish_tracked_pose`](https://google-cartographer-ros.readthedocs.io/en/latest/configuration.html) it seems it gives an error message as if it doesn't recognize those parameters.

I'll try to use a newer version and see how it goes.

Otherwise the mapping seems to work quite well.

moooeeeep, 2021-09-15:
After doing some more research I found a configuration where it would publish the transforms.

I let my odometry source publish the odometry transform (instead of just providing the topic) and adjusted the lua config basically to what you have too:

    ...
    map_frame = "map",
    tracking_frame = "imu_frame",
    published_frame = "odom",
    odom_frame = "odom",
    provide_odom_frame = false,
    ...

Now it publishes the transform between map and odom, the odometry source publishes base_link to odom, and so I have the tree I always wanted.

Not sure if this was actually "the same issue" in the end.

bharath5673, 2023-01-28:
**this will help**

```
sudo apt install ros-humble-gazebo-ros
sudo apt-get install ros-humble-gazebo-msgs
sudo apt-get install ros-humble-gazebo-plugins
```