cartographer-project/cartographer_ros #1674: Intra-submap constraint between SubmapID = (2, 40) and NodeID (2, 1847) is missing submap data

GrantZheng86, 2021-09-09:
Hello,

I've tried to perform map update through #1595, and it looks that "overlapping submap trimmer" is doing its job at updating the maps in RVIZ. 

However, there have always been issues like "missing sub map data" (the same at the title) , and crashes like ```
"terminate called after throwing an instance of 'std::out_of_range'
  what():  map::at"
```

I ran the bag validation and it looks like everything is fine, except some numbers are off for the IMU data. But I have it turned off for cartographer. Here's the code output from validation.

```
W0908 23:31:56.840755 29693 rosbag_validate_main.cc:103] frame_id os_imu time 1630987825011531820: IMU linear acceleration is 2.91371 m/s^2, expected is [3, 30] m/s^2. (It should include gravity and be given in m/s^2.) linear_acceleration -0.244209 0-1.04148 002.71024
W0908 23:31:56.962929 29693 rosbag_validate_main.cc:103] frame_id os_imu time 1630988380325374180: IMU linear acceleration is 2.04012 m/s^2, expected is [3, 30] m/s^2. (It should include gravity and be given in m/s^2.) linear_acceleration -1.03669 0.962469 01.47004
W0908 23:31:56.962975 29693 rosbag_validate_main.cc:103] frame_id os_imu time 1630988380335381495: IMU linear acceleration is 1.42674 m/s^2, expected is [3, 30] m/s^2. (It should include gravity and be given in m/s^2.) linear_acceleration -0.356736 00.280122 001.35272
I0908 23:31:57.003327 29693 rosbag_validate_main.cc:399] Time delta histogram for consecutive messages on topic "/os_cloud_node/imu" (frame_id: "os_imu"):
Count: 101457  Min: 1.3127e-05  Max: 0.0245041  Mean: 0.0100018
[0.000013, 0.002462)	                    	Count: 30 (0.0295692%)	Total: 30 (0.0295692%)
[0.002462, 0.004911)	                    	Count: 130 (0.128133%)	Total: 160 (0.157702%)
[0.004911, 0.007360)	                    	Count: 967 (0.953113%)	Total: 1127 (1.11082%)
[0.007360, 0.009810)	                  ##	Count: 11832 (11.6621%)	Total: 12959 (12.7729%)
[0.009810, 0.012259)	   #################	Count: 87005 (85.7555%)	Total: 99964 (98.5284%)
[0.012259, 0.014708)	                    	Count: 1267 (1.2488%)	Total: 101231 (99.7772%)
[0.014708, 0.017157)	                    	Count: 192 (0.189243%)	Total: 101423 (99.9665%)
[0.017157, 0.019606)	                    	Count: 27 (0.0266123%)	Total: 101450 (99.9931%)
[0.019606, 0.022055)	                    	Count: 2 (0.00197128%)	Total: 101452 (99.9951%)
[0.022055, 0.024504]	                    	Count: 5 (0.0049282%)	Total: 101457 (100%)
E0908 23:31:57.003893 29693 rosbag_validate_main.cc:383] Point data (frame_id: "os_sensor") has a large gap, largest is 0.127746 s, recommended is [0.0005, 0.05] s with no jitter.
I0908 23:31:57.004065 29693 rosbag_validate_main.cc:399] Time delta histogram for consecutive messages on topic "/reduced_points" (frame_id: "os_sensor"):
Count: 10146  Min: 0.0645728  Max: 0.127746  Mean: 0.100009
[0.064573, 0.070890)	                    	Count: 1 (0.0098561%)	Total: 1 (0.0098561%)
[0.070890, 0.077207)	                    	Count: 11 (0.108417%)	Total: 12 (0.118273%)
[0.077207, 0.083525)	             #######	Count: 3420 (33.7079%)	Total: 3432 (33.8261%)
[0.083525, 0.089842)	                 ###	Count: 1405 (13.8478%)	Total: 4837 (47.674%)
[0.089842, 0.096159)	                    	Count: 16 (0.157698%)	Total: 4853 (47.8317%)
[0.096159, 0.102477)	                  ##	Count: 1135 (11.1867%)	Total: 5988 (59.0183%)
[0.102477, 0.108794)	                    	Count: 5 (0.0492805%)	Total: 5993 (59.0676%)
[0.108794, 0.115111)	                    	Count: 52 (0.512517%)	Total: 6045 (59.5801%)
[0.115111, 0.121428)	                 ###	Count: 1755 (17.2975%)	Total: 7800 (76.8776%)
```

Any suggestions are greatly appreciated!

Thanks, everyone!

GrantZheng86, 2021-10-14:
@imstevenpmwyca Thanks a lot for the suggestion!

Regarding the "missing submap data", I have not found a way to deal with that. Currently, I am only tuning my parameters to lower the latency to alleviate the problem. Whenever that warning message popped up, my localization will freeze for a couple of seconds. By tuning for lower latency, the program should be able to find the correct location more quickly (and reduce drifting) after it gets out of the freezing issue.

I am still actively searching for the cause, and I'll keep you updated.

GrantZheng86, 2021-10-15:
I was basically following [this doc](https://google-cartographer-ros.readthedocs.io/en/latest/tuning.html#low-latency). The most significant impact on speed for my application is the resolution and "optimize every n node." I reduced the default 0.05 to 0.07, and my program optimized every 20-30 nodes. Lowering the sampling ratios to 1/10 of the original value can still have a good localization performance.

All these values could vary significantly based on your application. I am localizing in a medium-sized warehouse.

GrantZheng86, 2021-12-13:
Hello @imstevenpmwyca,

I've got some new discoveries: 
The "missing submap data" problem seems to go away when I bumped the "min_covered_area" a lot higher. I had it to be more than 50 while the default value was only 2. My "TRAJECTORY_BUILDER_2D.submaps.num_range_data" was set to 50 (indicating the submap size). I turned on the submap visualization in RVIZ, basically any "min_covered_area" lower than 30 doesn't seem to do anything in updating my maps.

imstevenpmwyca, 2021-12-14:
Dear @GrantZheng86,

Thanks for the information. However, by increasing the 'min_covered_area' parameters, you are using a *very* conservative trimmer. That makes me think that maybe you are not getting the error anymore because you are barely trimming any submaps. Have you check the number of nodes and constraints before and after trimming? Just to make sure the trimmer is being used.

GrantZheng86, 2021-12-14:
@imstevenpmwyca 

I visualized the trimmer in action by turning on "submap" topic in RVIZ, new submaps will show up in real-time, and you can see old submaps get deleted from the list. I haven't directly checked the constraints and nodes. My current theory is that constraints and nodes should be inside of submaps (which could be totally wrong). As submaps get trimmed, those should also be trimmed altogether (Again, not 100% sure if this is correct).  

Through my experiment, I saw that the higher the "min_covered_area" is, the faster some old submaps will get trimmed, and I got a lot fewer submaps to end with compared with low "min_covered_area". For a too high value of "min_covered_area", I often get an over-trimmed map, meaning some parts of my map are missing. 

Also, I think lines 26 and 27 in [this](https://github.com/cartographer-project/cartographer/blob/master/cartographer/mapping/internal/2d/overlapping_submaps_trimmer_2d.h) explain things well.

imstevenpmwyca, 2021-12-14:
Dear @GrantZheng86 ,

My bad! Yes you are right, the higher the parameter is, the looser the trimmer is. Please ignore my last comment. But then I don't really get how is this solving the 'missing submap data' issue. Maybe that could gives us an idea of what is really failing under the hood.

Also, I believe you assumptions are correct. The submaps are a result of the nodes and constraints so they are directly proportional to each other. And yes, it is expected that with such parameters, your map loses information to a great extent.