cartographer-project/cartographer_ros #1676: errors in step "catkin_make_isolated --install --use-ninja"

BonghanKim, 2021-09-27:
Hi,
I am having the following error when running `catkin_make_isolated --install --use-ninja':

Failed to process package 'cartographer':
Command '['ninja', '-j16', '-l16']' returned non-zero exit status 1

I was following this documentation:
https://google-cartographer-ros.readthedocs.io/en/latest/compilation.html

I am running Ubunut 16.04 with ROS kinetic.

bongbong@bongbong-GP66-Leopard-11UG:~/carto_ws$ catkin_make_isolated --install --use-ninja
Base path: /home/bongbong/carto_ws
Source space: /home/bongbong/carto_ws/src
Build space: /home/bongbong/carto_ws/build_isolated
Devel space: /home/bongbong/carto_ws/devel_isolated
Install space: /home/bongbong/carto_ws/install_isolated

~~  traversing 4 packages in topological order:
~~  - cartographer (plain cmake)
~~  - cartographer_ros_msgs
~~  - cartographer_ros
~~  - cartographer_rviz

==> Processing plain cmake package: 'cartographer'
build.ninja exists, skipping explicit cmake invocation...
==> ninja build.ninja in '/home/bongbong/carto_ws/build_isolated/cartographer/install'
ninja: no work to do.
==> ninja -j16 -l16 in '/home/bongbong/carto_ws/build_isolated/cartographer/install'
[16/385] Running C++ protocol buffer c...graph/constraint_builder_options.proto
FAILED: cd /home/bongbong/carto_ws/build_isolated/cartographer/install && /usr/bin/protoc --cpp_out /home/bongbong/carto_ws/build_isolated/cartographer/install -I /home/bongbong/carto_ws/src/cartographer /home/bongbong/carto_ws/src/cartographer/cartographer/mapping/proto/pose_graph/constraint_builder_options.proto
cartographer/mapping/proto/pose_graph/constraint_builder_options.proto:15:10: Unrecognized syntax identifier "proto3". This parser only recognizes "proto2".
[16/385] Running C++ protocol buffer c...ng/ceres_scan_matcher_options_2d.proto
FAILED: cd /home/bongbong/carto_ws/build_isolated/cartographer/install && /usr/bin/protoc --cpp_out /home/bongbong/carto_ws/build_isolated/cartographer/install -I /home/bongbong/carto_ws/src/cartographer /home/bongbong/carto_ws/src/cartographer/cartographer/mapping/proto/scan_matching/ceres_scan_matcher_options_2d.proto
cartographer/mapping/proto/scan_matching/ceres_scan_matcher_options_2d.proto:15:10: Unrecognized syntax identifier "proto3". This parser only recognizes "proto2".
[16/385] Running C++ protocol buffer c...ng/ceres_scan_matcher_options_3d.proto
FAILED: cd /home/bongbong/carto_ws/build_isolated/cartographer/install && /usr/bin/protoc --cpp_out /home/bongbong/carto_ws/build_isolated/cartographer/install -I /home/bongbong/carto_ws/src/cartographer /home/bongbong/carto_ws/src/cartographer/cartographer/mapping/proto/scan_matching/ceres_scan_matcher_options_3d.proto
cartographer/mapping/proto/scan_matching/ceres_scan_matcher_options_3d.proto:15:10: Unrecognized syntax identifier "proto3". This parser only recognizes "proto2".
[16/385] Running C++ protocol buffer c...pping/proto/submap_visualization.proto
FAILED: cd /home/bongbong/carto_ws/build_isolated/cartographer/install && /usr/bin/protoc --cpp_out /home/bongbong/carto_ws/build_isolated/cartographer/install -I /home/bongbong/carto_ws/src/cartographer /home/bongbong/carto_ws/src/cartographer/cartographer/mapping/proto/submap_visualization.proto
cartographer/mapping/proto/submap_visualization.proto:15:10: Unrecognized syntax identifier "proto3". This parser only recognizes "proto2".
[16/385] Running C++ protocol buffer c...relative_scan_matcher_options_3d.proto
FAILED: cd /home/bongbong/carto_ws/build_isolated/cartographer/install && /usr/bin/protoc --cpp_out /home/bongbong/carto_ws/build_isolated/cartographer/install -I /home/bongbong/carto_ws/src/cartographer /home/bongbong/carto_ws/src/cartographer/cartographer/mapping/proto/scan_matching/fast_correlative_scan_matcher_options_3d.proto
cartographer/mapping/proto/scan_matching/fast_correlative_scan_matcher_options_3d.proto:15:10: Unrecognized syntax identifier "proto3". This parser only recognizes "proto2".
[16/385] Running C++ protocol buffer c...correlative_scan_matcher_options.proto
FAILED: cd /home/bongbong/carto_ws/build_isolated/cartographer/install && /usr/bin/protoc --cpp_out /home/bongbong/carto_ws/build_isolated/cartographer/install -I /home/bongbong/carto_ws/src/cartographer /home/bongbong/carto_ws/src/cartographer/cartographer/mapping/proto/scan_matching/real_time_correlative_scan_matcher_options.proto
cartographer/mapping/proto/scan_matching/real_time_correlative_scan_matcher_options.proto:15:10: Unrecognized syntax identifier "proto3". This parser only recognizes "proto2".
[16/385] Running C++ protocol buffer c...pping/proto/connected_components.proto
FAILED: cd /home/bongbong/carto_ws/build_isolated/cartographer/install && /usr/bin/protoc --cpp_out /home/bongbong/carto_ws/build_isolated/cartographer/install -I /home/bongbong/carto_ws/src/cartographer /home/bongbong/carto_ws/src/cartographer/cartographer/mapping/proto/connected_components.proto
cartographer/mapping/proto/connected_components.proto:15:10: Unrecognized syntax identifier "proto3". This parser only recognizes "proto2".
[16/385] Running C++ protocol buffer c...f_range_data_inserter_options_2d.proto
FAILED: cd /home/bongbong/carto_ws/build_isolated/cartographer/install && /usr/bin/protoc --cpp_out /home/bongbong/carto_ws/build_isolated/cartographer/install -I /home/bongbong/carto_ws/src/cartographer /home/bongbong/carto_ws/src/cartographer/cartographer/mapping/proto/tsdf_range_data_inserter_options_2d.proto
cartographer/mapping/proto/tsdf_range_data_inserter_options_2d.proto:15:10: Unrecognized syntax identifier "proto3". This parser only recognizes "proto2".
[16/385] Running C++ protocol buffer c...mapping/proto/submaps_options_2d.proto
FAILED: cd /home/bongbong/carto_ws/build_isolated/cartographer/install && /usr/bin/protoc --cpp_out /home/bongbong/carto_ws/build_isolated/cartographer/install -I /home/bongbong/carto_ws/src/cartographer /home/bongbong/carto_ws/src/cartographer/cartographer/mapping/proto/submaps_options_2d.proto
cartographer/mapping/proto/submaps_options_2d.proto:15:10: Unrecognized syntax identifier "proto3". This parser only recognizes "proto2".
[16/385] Running C++ protocol buffer c...grapher/mapping/proto/pose_graph.proto
FAILED: cd /home/bongbong/carto_ws/build_isolated/cartographer/install && /usr/bin/protoc --cpp_out /home/bongbong/carto_ws/build_isolated/cartographer/install -I /home/bongbong/carto_ws/src/cartographer /home/bongbong/carto_ws/src/cartographer/cartographer/mapping/proto/pose_graph.proto
cartographer/mapping/proto/pose_graph.proto:15:10: Unrecognized syntax identifier "proto3". This parser only recognizes "proto2".
[16/385] Running C++ protocol buffer c...rtographer/mapping/proto/tsdf_2d.proto
FAILED: cd /home/bongbong/carto_ws/build_isolated/cartographer/install && /usr/bin/protoc --cpp_out /home/bongbong/carto_ws/build_isolated/cartographer/install -I /home/bongbong/carto_ws/src/cartographer /home/bongbong/carto_ws/src/cartographer/cartographer/mapping/proto/tsdf_2d.proto
cartographer/mapping/proto/tsdf_2d.proto:15:10: Unrecognized syntax identifier "proto3". This parser only recognizes "proto2".
[16/385] Running C++ protocol buffer c...her/mapping/proto/cell_limits_2d.proto
FAILED: cd /home/bongbong/carto_ws/build_isolated/cartographer/install && /usr/bin/protoc --cpp_out /home/bongbong/carto_ws/build_isolated/cartographer/install -I /home/bongbong/carto_ws/src/cartographer /home/bongbong/carto_ws/src/cartographer/cartographer/mapping/proto/cell_limits_2d.proto
cartographer/mapping/proto/cell_limits_2d.proto:15:10: Unrecognized syntax identifier "proto3". This parser only recognizes "proto2".
[16/385] Running C++ protocol buffer c...mapping/proto/submaps_options_3d.proto
FAILED: cd /home/bongbong/carto_ws/build_isolated/cartographer/install && /usr/bin/protoc --cpp_out /home/bongbong/carto_ws/build_isolated/cartographer/install -I /home/bongbong/carto_ws/src/cartographer /home/bongbong/carto_ws/src/cartographer/cartographer/mapping/proto/submaps_options_3d.proto
cartographer/mapping/proto/submaps_options_3d.proto:15:10: Unrecognized syntax identifier "proto3". This parser only recognizes "proto2".
[16/385] Running C++ protocol buffer c...rapher/mapping/proto/hybrid_grid.proto
FAILED: cd /home/bongbong/carto_ws/build_isolated/cartographer/install && /usr/bin/protoc --cpp_out /home/bongbong/carto_ws/build_isolated/cartographer/install -I /home/bongbong/carto_ws/src/cartographer /home/bongbong/carto_ws/src/cartographer/cartographer/mapping/proto/hybrid_grid.proto
[... cut ...]
cartographer/mapping/proto/scan_matching/fast_correlative_scan_matcher_options_2d.proto:15:10: Unrecognized syntax identifier "proto3". This parser only recognizes "proto2".
[16/385] cd /home/bongbong/carto_ws/bu...d/cartographer/install/AllFiles.cmake"
ninja: build stopped: subcommand failed.
<== Failed to process package 'cartographer':
Command '['ninja', '-j16', '-l16']' returned non-zero exit status 1

Reproduce this error by running:
==> cd /home/bongbong/carto_ws/build_isolated/cartographer && ninja -j16 -l16

Command failed, exiting.

vargabalint92, 2021-11-10:
Hi,

I have the same error

W-E-A, 2023-03-12:
The error message indicates that you are using proto2 as the protocol buffer version, but your code contains proto3 syntax. To resolve this, you can either modify your code to use proto2 syntax or upgrade your protocol buffer version to proto3. If you choose to upgrade to proto3, you need to declare the syntax in your code as follows: "syntax = 'proto3';". If you prefer to use proto2 syntax, you can declare it as "syntax = 'proto2';". However, if you upgrade to proto3, you need to ensure that all syntax in your code is compliant with proto3 standards.