cartographer-project/cartographer_ros #1678: How to properly use overlapping_submaps_trimmer_2d?

silasalves, 2021-09-29:
We are currently using Cartographer with SLAM on a small test environment. We observed that Cartographer's performance degraded over time and we concluded that the reason was the increasing number of submaps.

After a while, I found the [undocumented overlapping_submaps_trimmer_2d](https://google-cartographer.readthedocs.io/en/latest/configuration.html#cartographer-mapping-proto-mapbuilderoptions) and copied the configuration found in [pose_graph.lua](https://github.com/cartographer-project/cartographer/blob/master/configuration_files/pose_graph.lua#L90-L94). This is the configuration we used:

```lua
POSE_GRAPH.overlapping_submaps_trimmer_2d = {
  fresh_submaps_count = 1,
  min_covered_area = 2,
  min_added_submaps_count = 5,
}
```

This solved the performance problem, but created a new one: sometimes, the trimming process will decrease the certainty of obstacles (e.g., a wall may become free space) or completely "forget" parts of the map. This leads me to two major questions:

1. Is Cartographer intended to be used in SLAM mode all the time, or should I use the SLAM mode to map the environment first and then use pure localization?
2. What are the meaning of the following parameters: `fresh_submaps_count`, `min_covered_area` and `min_added_submaps_count`? How can I tune them so that the trimming process won't lose map data?

The full configuration file is as follows:

```lua
include "map_builder.lua"
include "trajectory_builder.lua"

options = {
  map_builder = MAP_BUILDER,
  trajectory_builder = TRAJECTORY_BUILDER,
  map_frame = "map",
  tracking_frame = "base_footprint",
  published_frame = "odom",
  odom_frame = "odom",
  provide_odom_frame = false,
  publish_frame_projected_to_2d = false,
  use_odometry = true,
  use_nav_sat = false,
  use_landmarks = true,
  num_laser_scans = 1,
  num_multi_echo_laser_scans = 0,
  num_subdivisions_per_laser_scan = 1,
  num_point_clouds = 0,
  lookup_transform_timeout_sec = 0.2,
  submap_publish_period_sec = 0.3,
  pose_publish_period_sec = 5e-3,
  trajectory_publish_period_sec = 30e-3,
  rangefinder_sampling_ratio = 1.,
  odometry_sampling_ratio = 1.,
  fixed_frame_pose_sampling_ratio = 1.,
  imu_sampling_ratio = 1.,
  landmarks_sampling_ratio = 1.,
}

MAP_BUILDER.use_trajectory_builder_2d = true
TRAJECTORY_BUILDER.collate_landmarks = false

TRAJECTORY_BUILDER_2D.min_range = 0.15
TRAJECTORY_BUILDER_2D.max_range = 3.
TRAJECTORY_BUILDER_2D.missing_data_ray_length = 1.
TRAJECTORY_BUILDER_2D.use_imu_data = true
TRAJECTORY_BUILDER_2D.use_online_correlative_scan_matching = true
TRAJECTORY_BUILDER_2D.motion_filter.max_angle_radians = math.rad(0.1)

POSE_GRAPH.constraint_builder.min_score = 0.65
POSE_GRAPH.constraint_builder.global_localization_min_score = 0.7
POSE_GRAPH.overlapping_submaps_trimmer_2d = {
  fresh_submaps_count = 1,
  min_covered_area = 2,
  min_added_submaps_count = 5,
}

return options
```

Thanks a lot for your help!

GrantZheng86, 2021-10-06:
I know there's pure_localization_trimmer specifically for localization only. I believe the overlapping trimmer is used for map update, and it looks like there might be a bug in the trimming code. I kept getting out of range error when I turned the overlapping trimmer on.

imstevenpmwyca, 2021-11-19:
Hello!
The vast majority of posegraph SLAM approaches have an problem: the longer you map (and some other factors I'm omitting for simplicity), the more complex the underlying posegraph becomes and hence the algorithm's performance degrades over time. This is a known issue and therefore, several approaches to how to bound the complexity of the posegraph have been developed. However, implementing so comes usually with a price: you have to delete some portions of it. If you want to continue SLAM-ing multiple and for long times while also getting a good performance, then you will need to -aggressively- trim the posegraph more often, risking deletion of data that was needed/useful for the whole process. How this deletion or trimming works is the main difference between bounding approaches.  

The one implemented in cartographer is indeed the one that you found, the `overlapping_submaps_trimmer_2d` (for 2D SLAM). It is a very simple approach, it favors the most recent data added to the posegraph over the out-dated one (as form of a submap). If the old submaps don't have enough information that the most recent already have, then the old submap is deleted. That is, if a sumap is at some degree overlapped by any number of newer ones, then we might consider its deletion. The degree of overlapping is reflected in the parameter `min_covered_area`, the number of newer submaps on top to be kept is reflected in `fresh_submaps_count` and finally the `min_added_submaps_count` is just a threshold to decide whether we consider using the trimmer in that area or not. However, keep in mind that the most conservative the parameters of the trimmer are, the less data will be deleted but then, the closer you will get to an under-performing algorithm. On the other hand, the most loose the trimmer is, the best bounded the posegraph will be and the best performance you will get but, the most data will be deleted.

So to answer precisely your questions:
1. `Is Cartographer intended to be used in SLAM mode all the time?` No, but that doesn't mean it can't... but you probably shouldn't. `Should I use the SLAM mode to map the environment first and then use pure localization?` That sounds reasonable, but it comes down to your specific application and intents.
2. `What are the meaning of the following parameters: fresh_submaps_count, min_covered_area and min_added_submaps_count?` Hopefully this is clear from the explanation above, but if it isn't you can [check the comments in the src file](https://github.com/cartographer-project/cartographer/blob/master/cartographer/mapping/internal/2d/overlapping_submaps_trimmer_2d.h). `How can I tune them so that the trimming process won't lose map data?` Hopefully you now have a better intuition about this one: the use of the trimmer will always imply deletion of map information but it is up to you to find right balance for your needs.

Hope this helps! Please consider closing the issue.

silasalves, 2021-11-22:
Thanks, @imstevenpmwyca, for the throughout explanation! It's much more clear to me how Cartographer handles the trimming process.

We are moving to map the environment first (probably using the cartographer_offline node) then using pure localization, as our application allows doing so.

Thanks a lot!

closed 2021-11-22 by silasalves