cartographer-project/cartographer_ros #1679: Poor scan matching

yuhui-xie, 2021-09-30:
1. scan match fail at the beginning, plz help me  o(╥﹏╥)o.
2.  https://gist.github.com/yuhui-xie/2a11b46c867d5e14d2dbed610782f0a8
3. https://github.com/yuhui-xie/carto_tunning, I didn't modify the source code, so only the launch file and corresponding lua file are upload.
4. https://drive.google.com/file/d/1gBaprWD40OgutrFuF7h6D1_kKEX96tkT/view?usp=sharing
the scan I use is sick 571

eithwa, 2021-10-04:
Unable to use bag file, access permission required.

yuhui-xie, 2021-10-04:
> Unable to use bag file, access permission required.

you can not play the bag file, or download the bag file?

eithwa, 2021-10-04:
> > Unable to use bag file, access permission required.
> 
> you can not play the bag file, or download the bag file?

[image]

yuhui-xie, 2021-10-04:
> > > Unable to use bag file, access permission required.
> > 
> > 
> > you can not play the bag file, or download the bag file?
> 
> [image]

sorry, I forget to adjust the opinion. I make a new one
https://drive.google.com/file/d/1gBaprWD40OgutrFuF7h6D1_kKEX96tkT/view?usp=sharing

eithwa, 2021-10-04:
[image]
I think there may be a problem with your odom, so I close it.
https://gist.github.com/eithwa/d6543f9cea98c5a7107913749ee0fba9

yuhui-xie, 2021-10-05:
> [image] I think there may be some problem with your odom, so I close it. https://gist.github.com/eithwa/d6543f9cea98c5a7107913749ee0fba9

wow thank you very much.
but I am wondering how do you find the odom doesn't work well?
I have draw the trajectory of it before, and found it looks like the way it have walked before exactly.

eithwa, 2021-10-05:
[image]
[image]

你的里程計Z軸數值不斷劇烈跳動
先把它修好可以解決部分問題