cartographer-project/cartographer_ros #1694: No tf transform between map and my own odom

Rikyuugo, 2021-11-18:
Hi, I am a new to cartographer and i have read different issues but still can't determine what caused it.
I am using ros-melodic and install the cartographer by sudo apt-get install ros-melodic-carotgrapher_ros.
And i have a 3D lidar ,imu on a real robot.
Now,I wrote the odometer file and broadcast the TF tree between base_link and odom.The frame_id of my lidar is laser_link,the frame_id of my imu is imu_link,and odom's frame_id is odom.
I modified the demo_ backpack_ 3d.launch,backpack_3d.launch and backpack_3d.lua as shown below.
[issue.zip](https://github.com/cartographer-project/cartographer_ros/files/7564269/issue.zip)
This is my tf_tree.
[image]
After rosrun tf static_tf_tranform 0 0 0 0 0 0 world map 100
[image]

At the same time ,when i roslaunch demo_backpack_3d.launch ,it seems to have a variety of situations.
A situation
[image]
Another situation（According to the current configuration, this situation occurs most）
[image]
Rviz shows this，and /scan_mached_point （I don't know what this is, and I don't seem to have posted this topic）shows the radar, but it seems only once，of course, there is no tf between map and Odom
[image]

I also tried to set provide_odom_frame = true  ,  use_odometry = false,instead of using my own odom,but there was still no transform(also there is no tf between base_link and odom)
What caused my problem, parameters or something？Or is there any way to build maps only by relying on lidar？Thanks a lot.

eithwa, 2021-11-19:
Can you record a bag file to provide testing?

Rikyuugo, 2021-11-19:
> can you record a bag file to provide testing? and the zip file you provided is missing the file "backpack_3d.launch"

I use rosbag rocord -a and rocord about 30s.I'm not sure if it meets your needs. If you have any questions, please continue to contact me
[https://github.com/Rikyuugo/issuebag.git](url)

eithwa, 2021-11-19:
[image]
File cannot be opened

eithwa, 2021-11-19:
imu topic does not look good, I think you need to confirm your hardware
[image]

Rikyuugo, 2021-11-19:
> imu topic does not look good [image]
> 
> It should look like this under normal conditions. [image]
> 
> I use 2d slam and disable imu data, the map looks good [image]

Thanks a lot.I will try to build 2D map and confirm IMU. Can you share your configuration？

Rikyuugo, 2021-11-19:
> imu topic does not look good [image]
> 
> It should look like this under normal conditions. [image]
> 
> I use 2d slam and disable imu data, the map looks good [image]

I tried 2D mapping and succeeded.It seems that it is indeed the problem of IMU. Thank you very much.close

closed 2021-11-19 by Rikyuugo