cartographer-project/cartographer_ros #1702: Errors while Installing in ROS Noetic.

adityasivaraj, 2021-12-21:
Processing catkin package: 'cartographer_ros'
==> Building with env: '/home/ad/catkin_ws/install_isolated/env.sh'
==> cmake /home/ad/catkin_ws/src/cartographer_ros/cartographer_ros -DCATKIN_DEVEL_PREFIX=/home/ad/catkin_ws/devel_isolated/cartographer_ros -DCMAKE_INSTALL_PREFIX=/home/ad/catkin_ws/install_isolated -G Ninja in '/home/ad/catkin_ws/build_isolated/cartographer_ros'
-- Found Boost: /usr/lib/x86_64-linux-gnu/cmake/Boost-1.71.0/BoostConfig.cmake (found version "1.71.0") found components: system iostreams 
-- Eigen found (include: /usr/include/eigen3, version: 3.3.7)
-- Found Boost: /usr/include (found suitable version "1.71.0", minimum required is "1.55.0") found components: system filesystem date_time iostreams regex 
-- looking for PCL_COMMON
-- Found installed version of Eigen: /usr/lib/cmake/eigen3
-- Found required Ceres dependency: Eigen version 3.3.7 in /usr/include/eigen3
-- Found required Ceres dependency: glog
-- Found installed version of gflags: /usr/lib/x86_64-linux-gnu/cmake/gflags
-- Detected gflags version: 2.2.2
-- Found required Ceres dependency: gflags
-- Found Ceres version: 1.14.0 installed in: /usr with components: [EigenSparse, SparseLinearAlgebraLibrary, LAPACK, SuiteSparse, CXSparse, SchurSpecializations, OpenMP, Multithreading]
-- Build type: Release
-- Using CATKIN_DEVEL_PREFIX: /home/ad/catkin_ws/devel_isolated/cartographer_ros
-- Using CMAKE_PREFIX_PATH: /home/ad/catkin_ws/install_isolated;/home/ad/catkin_ws/devel;/opt/ros/noetic
-- This workspace overlays: /home/ad/catkin_ws/install_isolated;/home/ad/catkin_ws/devel;/opt/ros/noetic
-- Found PythonInterp: /usr/bin/python3.8 (found suitable version "3.8.10", minimum required is "3") 
-- Using PYTHON_EXECUTABLE: /usr/bin/python3.8
-- Using Debian Python package layout
-- Using empy: /usr/lib/python3/dist-packages/em.py
-- Using CATKIN_ENABLE_TESTING: ON
-- Call enable_testing()
-- Using CATKIN_TEST_RESULTS_DIR: /home/ad/catkin_ws/build_isolated/cartographer_ros/test_results
-- Forcing gtest/gmock from source, though one was otherwise available.
-- Found gtest sources under '/usr/src/googletest': gtests will be built
-- Found gmock sources under '/usr/src/googletest': gmock will be built
-- Found PythonInterp: /usr/bin/python3.8 (found version "3.8.10") 
-- Using Python nosetests: /usr/bin/nosetests3
-- catkin 0.8.10
-- BUILD_SHARED_LIBS is on
-- Using these message generators: gencpp;geneus;genlisp;gennodejs;genpy
CMake Error: The following variables are used in this project, but they are set to NOTFOUND.
Please set them or make sure they are set and tested correctly in the CMake files:
GMOCK_LIBRARY
    linked by target "time_conversion_test" in directory /home/ad/catkin_ws/src/cartographer_ros/cartographer_ros
    linked by target "configuration_files_test" in directory /home/ad/catkin_ws/src/cartographer_ros/cartographer_ros
    linked by target "msg_conversion_test" in directory /home/ad/catkin_ws/src/cartographer_ros/cartographer_ros
    linked by target "metrics_test" in directory /home/ad/catkin_ws/src/cartographer_ros/cartographer_ros

-- Configuring incomplete, errors occurred!
See also "/home/ad/catkin_ws/build_isolated/cartographer_ros/CMakeFiles/CMakeOutput.log".
See also "/home/ad/catkin_ws/build_isolated/cartographer_ros/CMakeFiles/CMakeError.log".
<== Failed to process package 'cartographer_ros': 
  Command '['/home/ad/catkin_ws/install_isolated/env.sh', 'cmake', '/home/ad/catkin_ws/src/cartographer_ros/cartographer_ros', '-DCATKIN_DEVEL_PREFIX=/home/ad/catkin_ws/devel_isolated/cartographer_ros', '-DCMAKE_INSTALL_PREFIX=/home/ad/catkin_ws/install_isolated', '-G', 'Ninja']' returned non-zero exit status 1.

Reproduce this error by running:
==> cd /home/ad/catkin_ws/build_isolated/cartographer_ros && /home/ad/catkin_ws/install_isolated/env.sh cmake /home/ad/catkin_ws/src/cartographer_ros/cartographer_ros -DCATKIN_DEVEL_PREFIX=/home/ad/catkin_ws/devel_isolated/cartographer_ros -DCMAKE_INSTALL_PREFIX=/home/ad/catkin_ws/install_isolated -G Ninja

I have been trying to install Cartographer_ROS in a ROS Noetic environment in Ubuntu 20.04 64bit PC, and I am getting these errors, how do I fix it?

adityasivaraj, 2021-12-22:
I am not able to reproduce this error due to changes in the environment. Hence, I am closing this issue.

closed 2021-12-22 by adityasivaraj

Happneel, 2022-11-28:
@adityasivaraj I ran into the same issue, I found out it was because of my "multi-ros" environment, I had ROS1 Noetic and ROS2 Galactic. I simply uninstalled ROS2 and ROS1 and finally reinstalled ROS1 (Noetic) and it worked!

marcelinomalmeidan, 2022-11-29:
I'm also having this issue on ROS Noetic. I don't have ROS2 installed, so IDK what's the fix for me

W-E-A, 2023-03-12:
Run "sudo apt-get install libgmock-dev" works for me.