cartographer-project/cartographer_ros #1708: cartographer pure localization parameter tuning issue - sudden shifts of scan data even the robot is stopped

hadush-hailu-geb, 2022-02-01:
Cartographer config, tf free, bag file and recorded video(of rviz)
https://drive.google.com/drive/folders/12qOU4kMDxgGG0_A7VqNz0mASXjAvpHHU?usp=sharing

Sensor:
Velodyne vlp-16

I'm having trouble finding the optimal configuration of pure localization mode of cartographer. At first, when the cartographer  pure-localization mode is launched, the localization process is pretty fast and accurate. I do not see any strange behaviors as well. But, I move(using Joystick) the robot around for a bit, the 2D scan data(filtered from velodyne 3D point cloud) starts to make a sudden shift once every 5 or 6 seconds and it continues to behave that way even the robot comes to stop(as in the attached video). 

Difference in transformation between two consecutive map->base_link tf data. 
[image]

I have tried to add the parameter "MAP_BUILDER.pose_graph.constraint_builder.min_score = 0.7" as suggested [here](https://github.com/cartographer-project/cartographer_ros/issues/249), but could not fix.

Why is the scan-matching not working once in a few seconds?

Thank you for your cooperation.

hadush-hailu-geb, 2022-02-02:
help

hadush-hailu-geb, 2022-02-02:
Can you help me understand why tf  map--> laser is moving when the robot is static?

eithwa, 2022-04-12:
[image]
does this problem also occur when you use bag file, I tested your file but everything works fine